UtilitechAS/amsreader-firmware #641: Final testing of 2.2.22 before release

gskjold, 2023-09-21:
We have some bugfixes being released soon, and it would be good to have a couple of people test it for a few days before final release

- Fixed red HAN indicator caused by millisecond rollover at 49.7 days
- Fixed unexpected loss of graph data
- Fixed unexpected overwrite of tariff peaks
- Fixed incorrect calculation of realtime "income" from export
- Fixed fetching of incorrect prices if fetched the first hour after midnigt
- Fixed incorrect date on clock in GUI
- Fixed uptime rollover issue for Raw MQTT payload
- Fixed missing tariff peak in MQTT HA payload
- Improved graph y-axis range when negative values are present
- Improved WiFi disconnect to preserve power when connection is lost
- Added persistence to realtime data to make it survive a software reboot (does not survive hardware reboot)
- Added voltage monitoring and automatic WiFi disconnect on low voltage for Pow-U (To prevent brownout)
- Added support for Slovenian Iskra meter

[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12681855/esp32solo.zip)
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12681856/esp32.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12681857/esp32s2.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12681858/esp32c3.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12681859/esp8266.zip)

ThorAlex87, 2023-09-21:
I've downloaded and installed, so far it looks good! Graph scaling and current month peaks i HA are working, I'll report on export calculations once I've got a few hours of data in.

ThorAlex87, 2023-09-21:
So far the export data looks to be correct and my energy dashboard is starting to make sense!

bmork, 2023-09-22:
Hatt denne installert ca et døgn nå. Dessverre fremdeles uten skikkelig oppetidslogging.  Men jeg ser nå at den har restartet for en time siden med den velkjente "Reason: Vbat power on reset (1/0)", så det er ikke noe som tyder på bedring her.

ArnieO, 2023-09-22:
> Hatt denne installert ca et døgn nå. Dessverre fremdeles uten skikkelig oppetidslogging. Men jeg ser nå at den har restartet for en time siden med den velkjente "Reason: Vbat power on reset (1/0)", så det er ikke noe som tyder på bedring her.

I believe you have purchased both Pow-U and Pow-K. Which one is this?

bmork, 2023-09-22:
> I believe you have purchased both Pow-U and Pow-K. Which one is this?

The Pow-U is the only one having these issues.

The Pow-K is extremely stable and has been so for a long time and many releases.  It is now running v2.2.21 with 73 days uptime.  The last reboot reason was "Software restart (4/0)", and I guess that was when it was upgraded to that version.

ArnieO, 2023-09-22:
> The Pow-U is the only one having these issues.

OK - thank you for confirming.

pomok, 2023-09-23:
Test version installed. No reboots so far. Uptime now 20hrs.

gskjold, 2023-09-23:
@bmork thanks for the feedback, can you please confirm from the status page that it says "Device: Pow-U+" ? The changes are board profile specific

bmork, 2023-09-23:
It's a Pw-U+.  Status page says
```
Device information
Chip: esp32s2 (240MHz)
Device: [Pow-U+](http://ams.mork.no/vendor)
MAC: 58:CF:79:9B:44:BE
AP MAC: 58:CF:79:9B:44:BF
Last boot: 23.09.2023 11:30
Reason: Vbat power on reset (1/0)
```

gskjold, 2023-09-23:
Today I remembered that there was a PR from @dbeinder that enables you to disable 802.11b modes. Apparently preventing the device from dropping into these lower modes will help to preserve power. So, if the previous version is not working for you, please try this one and disable both of these options in config:
[image]
Unless you actually know you have a 11b-only network, which I doubt if you have fairly modern wifi...

[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12706412/esp32c3.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12706413/esp32s2.zip)
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12706414/esp32.zip)
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12706415/esp32solo.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12706416/esp8266.zip)

bmork, 2023-09-23:
done.  Will report back how it goes.

The closest AP should be modern enough.  Manufactured a couple of months ago, based on a MT7981 SoC, and running current+ OpenWrt master :-)

Here's the Pow-U+ associtation before and after this firmware upgrade in case it matters:
```
Station 58:cf:79:9b:44:be (on phy0-ap0)
        inactive time:  1010 ms
        rx bytes:       5011147
        rx packets:     42748
        tx bytes:       2539380
        tx packets:     32884
        tx retries:     1391
        tx failed:      1391
        rx drop misc:   3398
        signal:         -59 [-70, -59] dBm
        signal avg:     -61 [-72, -61] dBm
        tx bitrate:     65.0 MBit/s MCS 7
        tx duration:    31391349 us
        rx bitrate:     72.2 MBit/s MCS 7 short GI
        rx duration:    21284764 us
        last ack signal:-61 dBm
        avg ack signal: -60 dBm
        airtime weight: 256
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       short
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short preamble: yes
        connected time: 10803 seconds
        associated at [boottime]:       404768.739s
        associated at:  1695369452514 ms
        current time:   1695380254847 ms

Station 58:cf:79:9b:44:be (on phy0-ap0)
        inactive time:  780 ms
        rx bytes:       66152
        rx packets:     450
        tx bytes:       28413
        tx packets:     370
        tx retries:     8
        tx failed:      8
        rx drop misc:   16
        signal:         -53 [-61, -53] dBm
        signal avg:     -55 [-64, -55] dBm
        tx bitrate:     65.0 MBit/s MCS 7
        tx duration:    418630 us
        rx bitrate:     72.2 MBit/s MCS 7 short GI
        rx duration:    199852 us
        last ack signal:-54 dBm
        avg ack signal: -53 dBm
        airtime weight: 256
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       short
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short preamble: yes
        connected time: 64 seconds
        associated at [boottime]:       415802.005s
        associated at:  1695380485779 ms
        current time:   1695380549546 ms
```

pomok, 2023-09-23:
[AMS reader b9d6f16] - reboot after about 22 hours.
Device information
Chip: esp32s2 (240MHz)
Device: [Pow-U+]()
Last boot: 23.09.2023 11:27
Reason: Vbat power on reset (1/0)

Meter
Manufacturer: Kaifa
Model: MA105H2E

bmork, 2023-09-23:
The version with 11b rates disabled looked promising for a while. But unfortunately I just had another "vbat" reset.

pomok, 2023-09-24:
[AMS reader b9d6f16] - **Reboot after 16 hours - not on Vbat but on Software reset**

Device information
Chip: esp32s2 (240MHz)
Device: [Pow-U+]()
Last boot: 24.09.2023 03:41
Reason: Software reset (3/0)

Meter
Manufacturer: Kaifa
Model: MA105H2E
ID: 6970631402121189

KennethLavrsen, 2023-09-25:
Just installed it on my K+. Initial - it works. I had a wrong price displayed and after upgrade it is correct. Will report more tomorrow

bmork, 2023-09-26:
FYI:  My Pow-U+ has been running with external power the last 24 hours.  Without any resets, as expected.   Just wanted to verify that the Vbat reason is real, and it seems so.

This also kind of eliminates network instability as possible cause of voltage dips.  My AP reports  "Cconnected time: 85169 seconds" for the Pow-U+, matching when the external power was connected..

I guess the next test is a full 24 hour packet capture, trying to see if there are situations where the Pow-U+ ends up trying to (re-)transmit data for too long periods.  Don't know exactly how to anylyze that, though.  Tips are welcome.

ArnieO, 2023-09-26:
> I guess the next test is a full 24 hour packet capture, trying to see if there are situations where the Pow-U+ ends up trying to (re-)transmit data for too long periods. Don't know exactly how to anylyze that, though. Tips are welcome.

Awsome - and hope you find a way to do this!

wknutsen, 2023-09-26:
[AMS reader b9d6f16]

I have been running this version in 5 days and still have frequently reboots (yellow line in graph).
So far the best firmware i had highest uptime with is the first test firmware in #627 with an uptime with almost 8 days.

I also have some issues with realtime hour import (See red line in graph) and export that some times stop accumulating hourly. Issue is present in frontend and on MQTT. This is a new issue that was not present in 2.2.21 or first test firmware on vbat issue.
However realtime accumulated daily import (See blue line in graph) and export seems to work fine.

[image]

ArnieO, 2023-09-26:
Great visualisation - thank you!

gskjold, 2023-09-26:
I re-enabled 802.11b this morning around 08.30, and the voltage was clearly more stable with 802.11b disabled:
[image]

For those still affected after disabling 802.11b in the latest firmware from #641, there must be something else affecting your power issues, we just haven't figured out what that is yet.

Please also note that any traffic directed agains the device could keep it alive, this includes multicast traffic on network, fetching json from API or even ping.

Regarding the newly found issue with realtime hourly, I will look into that

ArnieO, 2023-09-26:
> this includes multicast traffic on network

Is broad-/multicast in the network still an issue on ESP32, if Power Saving is set to Maximum? 
Ref: https://amsleser.no/blog/post/25-pow-u-and-power-saving

gskjold, 2023-09-26:
> Is broad-/multicast in the network still an issue on ESP32, if Power Saving is set to Maximum

Although the conditions improve on max power save, my assumption is that some of that traffic will still affect it, as the AP is responsible for sending that data and has no understanding of what sleep mode the ESP i currently in. It is my understanding that max power saving puts the modem in a deeper sleep state when sleeping, but it would still have to process incoming traffic at the intervals it wakes up.

bmork, 2023-09-26:
Does this include IPv6 neighbour discovery etc? Har to avoid all multicast traffic without configuring a completely dedicated network..

gskjold, 2023-09-26:
v6 shouldn't happen all that often, so I doubt that would make much of an impact

bmork, 2023-09-26:
right.  I guess SSDP, ARP and mDNS is worse.  How much is too much? 

I have simplified my network to a single "iot" net work all sorts of devices with limited need for Internet access.  This includes both the Pow-U+ and things like printers, speakers, ampifiers etc.  The latter can be a bt chatty announcing themselves.

I'll configure a separate network for the Pow-U+ and see if that makes a difference as soon as I'm finished running the traffic  capture.

gskjold, 2023-09-26:
> How much is too much?

That is the big question, I have yet to find out

gskjold, 2023-09-26:
I found the new problem where realtime is not updated. New firmware attached (81dedcb):
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12730951/esp32s2.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12730953/esp32c3.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12730954/esp8266.zip)
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12730955/esp32solo.zip)
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12730958/esp32.zip)

KennethLavrsen, 2023-09-26:
Looks good. Time is right, date isright, prices load right also just after midnight incl a reboot

gskjold, 2023-09-27:
> Looks good. Time is right, date isright, prices load right also just after midnight incl a reboot

Excellent, thank you for testing. I am never really awake that late, so I only had my automated test to support me on this one. Good to have it confirmed

bmork, 2023-09-27:
Still don't know exactly how to analyze the timing of this capure, but I found part of the conversation stats interesting:
```
# tshark -qnr /mnt/storage/pcap/pow-u-plus-1695711147.pcap -z conv,tcp
================================================================================
TCP Conversations
Filter:<No Filter>
                                                           |       <-      | |       ->      | |     Total     |    Relative    |   Duration   |
                                                           | Frames  Bytes | | Frames  Bytes | | Frames  Bytes |      Start     |              |
192.168.15.9:63298         <-> 192.168.15.1:1883           182264 9,842 kB   182264 24 MB      364528 34 MB         0.000000000    112862.2217
192.168.10.186:44646       <-> 192.168.15.9:80                  5 1,044 bytes       7 660 bytes      12 1,704 bytes     1.261427260         0.2979
192.168.10.186:44650       <-> 192.168.15.9:80                  6 1,104 bytes       6 606 bytes      12 1,710 bytes     6.555633179         0.0453
192.168.10.186:33502       <-> 192.168.15.9:80                  5 1,044 bytes       7 660 bytes      12 1,704 bytes    11.608737414         0.2831
192.168.10.186:33526       <-> 192.168.15.9:80                  5 1,044 bytes       7 660 bytes      12 1,704 bytes    16.888975407         0.0995
192.168.10.186:49860       <-> 192.168.15.9:80                  6 1,104 bytes       6 606 bytes      12 1,710 bytes    21.988915863         0.3588
192.168.10.186:49884       <-> 192.168.15.9:80                  5 1,044 bytes       7 660 bytes      12 1,704 bytes    27.356846862         0.0686
192.168.10.186:52544       <-> 192.168.15.9:80                  5 1,044 bytes       7 660 bytes      12 1,704 bytes    32.421819675         0.0622
192.168.10.186:52548       <-> 192.168.15.9:80                  5 1,044 bytes       7 660 bytes      12 1,704 bytes    37.487870209         0.2168
192.168.10.186:55948       <-> 192.168.15.9:80                  5 1,044 bytes       7 660 bytes      12 1,704 bytes    42.710104022         0.2243
192.168.10.186:55964       <-> 192.168.15.9:80                  5 1,044 bytes       7 660 bytes      12 1,704 bytes    47.948834705         0.1874
192.168.10.186:55618       <-> 192.168.15.9:80                  5 1,044 bytes       7 660 bytes      12 1,704 bytes    53.150084089         0.2339
192.168.10.186:55632       <-> 192.168.15.9:80                  5 1,044 bytes       7 660 bytes      12 1,704 bytes    58.403456656         0.1809
192.168.10.186:37416       <-> 192.168.15.9:80                  5 1,044 bytes       7 660 bytes      12 1,704 bytes    63.637025950         0.1757
192.168.10.186:37426       <-> 192.168.15.9:80                  5 1,044 bytes       7 660 bytes      12 1,704 bytes    68.848857162         0.1844
192.168.10.186:48626       <-> 192.168.15.9:80                  5 1,044 bytes       7 660 bytes      12 1,704 bytes    74.035664661         0.0317
192.168.10.186:48640       <-> 192.168.15.9:80                  5 1,044 bytes       7 660 bytes      12 1,704 bytes    79.264879053         0.2113
192.168.15.9:63301         <-> 13.49.120.40:80                  4 507 bytes       5 433 bytes       9 940 bytes  17069.703331125     86400.0189
192.168.15.9:63302         <-> 13.49.120.40:80                  4 327 bytes       4 373 bytes       8 700 bytes  103469.705715008         0.0643
192.168.10.186:33512       <-> 192.168.15.9:80                  3 180 bytes       4 236 bytes       7 416 bytes    11.859251554         5.0835
192.168.10.186:49876       <-> 192.168.15.9:80                  3 180 bytes       4 236 bytes       7 416 bytes    22.238988514         5.1466
192.168.15.9:63300         <-> 13.49.120.40:80                  0 0 bytes         1 60 bytes        1 60 bytes   17069.720160327         0.0000
================================================================================
```

The Pow-U+ is 192.168.15.9.  Ignore those 192.168.10.186 http accesses.  It took som time before I remembered to close all browser windows.

The sessions to 13.49.120.40 are related to the price API.  What really caught my eye is the one lasting exactly 24 hours.  That's pretty strange, isn't it?  Looks like the stats are right - the HTTP request finishes, but the tcp session isn't closed until a day later:

```
# tshark -nr /mnt/storage/pcap/pow-u-plus-1695711147.pcap -Y 'tcp.port==63301'
56057 17069.703331125 192.168.15.9 → 13.49.120.40 TCP 60 63301 → 80 [SYN] Seq=0 Win=5744 Len=0 MSS=1436
56058 17069.713839726 13.49.120.40 → 192.168.15.9 TCP 58 80 → 63301 [SYN, ACK] Seq=0 Ack=1 Win=26883 Len=0 MSS=1460
56059 17069.719072419 192.168.15.9 → 13.49.120.40 TCP 60 63301 → 80 [ACK] Seq=1 Ack=1 Win=5744 Len=0
56061 17069.725347495 192.168.15.9 → 13.49.120.40 HTTP 193 GET /hub/price/10YNO-1--------2/2023/9/27?currency=NOK HTTP/1.0 
56062 17069.735589933 13.49.120.40 → 192.168.15.9 TCP 54 80 → 63301 [ACK] Seq=1 Ack=140 Win=27872 Len=0
56063 17069.748663002 13.49.120.40 → 192.168.15.9 HTTP 341 HTTP/1.1 200  
56064 17069.748698479 13.49.120.40 → 192.168.15.9 TCP 54 80 → 63301 [FIN, ACK] Seq=288 Ack=140 Win=27872 Len=0
56065 17069.752198372 192.168.15.9 → 13.49.120.40 TCP 60 63301 → 80 [ACK] Seq=140 Ack=289 Win=5456 Len=0
341627 103469.722186925 192.168.15.9 → 13.49.120.40 TCP 60 63301 → 80 [FIN, ACK] Seq=140 Ack=289 Win=5456 Len=0
```

Not sure this is relevant, but interesting anyway.

Miklagarur, 2023-09-28:
The commitment and support on this device are well beyond expectations, as this thread clearly show, thank you. 
Latest version 81dedcb has been the most stable of them all and the device are running as expected on HAN port power only. Though today there is a minor issue, after midnight the Price API stopped (ENTSO-E) and are now showing blank. Tried to disable and re-enter the key without any luck. Not able to test if POW-U issue or ENTSO-E issue though. Anyone else experiencing this?

bmork, 2023-09-28:
> I'll configure a separate network for the Pow-U+

The Pow-U+ is now on a VLAN/SSID where a combined gateway, DNS server, DHCP server and MQTT broker is the only oher IP device.  The gateway does only limited forwarding, without any multicast forwarding or relaying. No global IPv6 addresses configured, but I haven't bothered disabling link local IPv6. The Pow-U+ is publshing MQTT but has no HTTP clients.

Still, it rebooted after 37207 seconds in this environment. So no go. I believe we can rule out unrelated multicast traffic, as the primary cause at least.

Just for the fun of it, I'm going to disable MQTT and see if it stays up without any traffic (looking at the AP association to know uptime)

Another test I also could do is switching from the current Wifi6 AP to a Wifi4 AP.  Don't know if that can make any difference?  I assume this device is limited to 802.11n anyway?

KennethLavrsen, 2023-09-28:
The night between 27th and 28th price fetching broke. No prices fetched. Does not load prices this morning the 28th after a reboot. Totally broken

ArnieO, 2023-09-28:
[Entso-E has not published data for 28th September.](https://transparency.entsoe.eu/transmission-domain/r2/dayAheadPrices/show?name=&defaultValue=false&viewType=GRAPH&areaType=BZN&atch=false&dateTime.dateTime=28.09.2023+00:00|CET|DAY&biddingZone.values=CTY|10YNO-0--------C!BZN|10YNO-3--------J&resolution.values=PT15M&resolution.values=PT30M&resolution.values=PT60M&dateTime.timezone=CET_CEST&dateTime.timezone_input=CET+(UTC+1)+/+CEST+(UTC+2))
[image]

gskjold, 2023-09-28:
Thank you @bmork for your extensive testing!

> That's pretty strange, isn't it? Looks like the stats are right - the HTTP request finishes, but the tcp session isn't closed until a day later

This is really interesting, I'm going to take a closer look at this!

> Still, it rebooted after 37207 seconds in this environment. So no go. I believe we can rule out unrelated multicast traffic, as the primary cause at least.

I agree, this makes traffic as the cause unlikely.

> Another test I also could do is switching from the current Wifi6 AP to a Wifi4 AP. Don't know if that can make any difference? I assume this device is limited to 802.11n anyway?

This is limited to 802.11n, yes, but I think the test would be valid still, as I have heard rumors that not all devices play well with wifi6. That being said, I have a device connected to a Unifi WiFi6 at one of my locations without issues. You had openwrt, right? What brand/type of APs are you running?

bmork, 2023-09-28:
> You had openwrt, right? What brand/type of APs are you running?

Ubiquiti all over the place at the moment.

Current AP is a Unifi 6 Plus (U6+).  Yes, running OpenWrt :-)

But I do have an old Unifi AP AC Pro within range too.  So I could switch to that one for a quick test of non-Wifi6 hardware.

wknutsen, 2023-09-28:
[AMS reader 81dedcb]

This version with disabled legacy rates seems very stable. No reboots so far.
[image]

ArnieO, 2023-09-28:
> This version with disabled legacy rates seems very stable. No reboots so far.

Sounds very promising! 🤞

bmork, 2023-09-28:
I don't think the problem can be caused by any IP traffic.  Just saw a reboot on the dedicated network with MQTT disabled and no HTTP clients.

This is a summary of the snoop I had running at the time (capturing everything on that network):  
```
 1123 28091.754341774 00:1b:21:a7:98:bc → 58:cf:79:9b:44:be ARP 42 Who has 192.168.17.9? Tell 192.168.17.1
 1124 28091.887153557 58:cf:79:9b:44:be → 00:1b:21:a7:98:bc ARP 60 192.168.17.9 is at 58:cf:79:9b:44:be
 1125 28146.686146737 58:cf:79:9b:44:be → ff:ff:ff:ff:ff:ff ARP 60 ARP Announcement for 192.168.17.9
 1126 28206.686680795 58:cf:79:9b:44:be → ff:ff:ff:ff:ff:ff ARP 60 ARP Announcement for 192.168.17.9
 1127 28221.750486347 192.168.17.9 → 192.36.143.130 NTP 90 NTP Version 4, client
 1128 28266.687240608 58:cf:79:9b:44:be → ff:ff:ff:ff:ff:ff ARP 60 ARP Announcement for 192.168.17.9
 1129 28284.092335250      0.0.0.0 → 255.255.255.255 DHCP 350 DHCP Discover - Transaction ID 0x97115c37
 1130 28284.092455788 192.168.17.1 → 192.168.17.9 DHCP 342 DHCP Offer    - Transaction ID 0x97115c37
 1131 28284.095366783      0.0.0.0 → 255.255.255.255 DHCP 350 DHCP Request  - Transaction ID 0x97115c37
 1132 28284.095454074 192.168.17.1 → 192.168.17.9 DHCP 342 DHCP ACK      - Transaction ID 0x97115c37
 1133 28284.130653252 58:cf:79:9b:44:be → ff:ff:ff:ff:ff:ff ARP 60 ARP Announcement for 192.168.17.9
 1134 28284.133708882 58:cf:79:9b:44:be → ff:ff:ff:ff:ff:ff ARP 60 Who has 192.168.17.1? Tell 192.168.17.9
 1135 28284.133715462 00:1b:21:a7:98:bc → 58:cf:79:9b:44:be ARP 42 192.168.17.1 is at 00:1b:21:a7:98:bc
 1136 28284.136727832 192.168.17.9 → 192.168.17.1 DNS 75 Standard query 0x9f6c A hub.amsleser.no
 1137 28284.181563132 192.168.17.1 → 192.168.17.9 DNS 146 Standard query response 0x9f6c A hub.amsleser.no CNAME ams-cloud.eu-north-1.elasticbeanstalk.com A 13.49.120.40
 1138 28284.917751666 192.168.17.9 → 192.168.17.1 DNS 72 Standard query 0x055b A pool.ntp.org
 1139 28284.940296727 192.168.17.1 → 192.168.17.9 DNS 136 Standard query response 0x055b A pool.ntp.org A 162.159.200.123 A 152.65.32.101 A 192.36.143.130 A 185.41.243.43
 1140 28284.946495308 192.168.17.9 → 162.159.200.123 NTP 90 NTP Version 4, client
```
Nothing exceptional.  It goes directly from an occasional ARP announcment and NTP query into reboot, shown as a DHCP transaction.

(BTW, the NTP queries are another bug I've meant to report for a long time:  Using DHCP assigned NTP servers does not work. Not shown here, but the DHCP reply configures 192.168.17.1 as NTP server so "pool.ntp.org" should not have been used)

KennethLavrsen, 2023-09-28:
Yes. Broken. But yes, Entso-E was broken. Problem repaired itself mid day

bmork, 2023-09-28:
And I just had a reboot while connected to the Wifi4 UAP AC Pro. Running out of stuff to try.  Guess I'll live with the reboots.  Better alternative than adding external power anyway.

gskjold, 2023-09-29:
Releasing this today. We will continue the voltage reboot problem in the original thread

closed 2023-09-29 by gskjold