UtilitechAS/amsreader-firmware #642: Translations for GUI

el97, 2023-09-22:
Hi! I have noticed that there is no option for different languages. I think it would be appreciated if the GUI supports more languages so more people can interact and understand how the energy is consumed.

**Describe the solution you'd like**
I would like to see an option for different languages. I'm willing to help out with a Swedish translation as well!

**Additional context**
Your work is fantastic and really helps me out when it comes to energy consumption.

ArnieO, 2023-09-22:
A good idea indeed! 
We have talked about this, but not prioritized it yet. Will be higher on the list once we get v2.3.x out (which is hopefully next week).
And implementing multi-language in a sufficiently flexible way - so that adding new languages is easy - will require a bit of time.

gskjold, 2024-01-20:
Here is a test version with translations. Please note:
- This is a development build. Only flash this if you know how to erase and flash official version. 
- Upgrading from a previous test version may cause problems, erase flash recommended. 
- Back up config before proceeding.
- Vanilla ESP32 image has become too large with the latest build and is not included here. Will fix before final release

Install firmware, go to config, open dropdown for "Language", choose "Load from server" and wait for it to update before selecting language and saving. 

Only English and Norwegian is available for now. If you fancy providing us with translations, we use Localazy: https://localazy.com/p/amsleser

[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13997991/esp32s2.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13997992/esp32c3.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13997993/esp8266.zip)

Noschvie, 2024-01-22:
Hello

Please remove "British English" because of a side effect adding "German"
[image]

Thanks!

ArnieO, 2024-01-22:
> Please remove "British English" because of a side effect adding "German"

Done!

Noschvie, 2024-01-22:
"German" added.

[image]
but have no device available for testing.

ArnieO, 2024-01-22:
> "German" added.
> 
> ![grafik](https://private-user-images.githubusercontent.com/48217146/298620695-b1649b9f-64d2-4649-a0fc-054b441ce5de.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU5MzQ3OTQsIm5iZiI6MTcwNTkzNDQ5NCwicGF0aCI6Ii80ODIxNzE0Ni8yOTg2MjA2OTUtYjE2NDliOWYtNjRkMi00NjQ5LWEwZmMtMDU0YjQ0MWNlNWRlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTIyVDE0NDEzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRjZTFmNWRjZTQzNWRlM2Y3YTNkZmQ4MTRjNWZjNGMyMzJjMmIwZDRkYzkyMDg0YWQ2M2ZmNWJkYTg2MTM2MjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.45oBulFloAHwRYp_Uj8H7mh5rGTjKK98YN9ihFiTJVA) but have no device available for testing.

Thank you!

ArnieO, 2024-01-23:
@Noschvie : Thank you for your effort in translation to German!
I have reviewed the translations, and approved all of them except a few. Could you be so kind as to reconsider these?
(I might have adjusted some of them while reviewing...)

```

Publish topic	-	Topic veröffentlichen	I think this is wrong
Upload cert	-	Zertifikat hochladen	Must be shorter. Click "SSL" in MQTT tile to see current button.
Upload key	-	Schlüssel hochladen	Must be shorter. Click "SSL" in MQTT tile to see current button.
Cert OK		-	Zertifikat OK		Should be shorter. Click "SSL" in MQTT tile to see current button.

Click here to upload certificate	-	Klicken Sie hier, um das Zertifikat hochzuladen	Propose shorter: "Hier klicken, um Zertifikat hochzuladen" (Repeated two times)

Click here to upload private key	-	Klicken Sie hier, um den privaten Schlüssel hochzuladen	Same as previous (Repeated two times)
Energy Speedometer	-	Energie-Tachometer	This is a product name, should not be translated.
Realtime plot	-	Echtzeit-Darstellung	Propose: Echtzeit diagramm
Month plot	-	Monatliche Darstellung	Propose: Monatliche diagramm
Current IDX	-	Aktueller IDX		Propose: Strom IDX
Factory reset	-	Zurücksetzen auf Werkseinstellungen	Too long
Payload boundry flag missing	-	Payload-Grentzwert fehlt	Propose: "Payload-Grenzflagge fehlt"
Debugging	-	Fehlerdiagnose	Propose: "Debuggen"
```

Noschvie, 2024-01-23:
Will have a look ... but need to install a test device.

ArnieO, 2024-01-23:
> @ArnieO please have a look to "Realtime" and "Real time" / without and with space in english

Will do, thank you!

Noschvie, 2024-01-24:
done and ready for review:

- "Current IDX" -> "Stromstärke IDX"
- "Publish topic" -> "Topic" (it's clear in the context of MQTT)

ArnieO, 2024-01-24:
> * "Current IDX" -> "Stromstärke IDX"
OK! 👍 

> * "Topic" (it's clear in the context of MQTT)
If we go for English here, I think we should use "Publish topic", ref what I wrote above?

ArnieO, 2024-01-29:
Y-a-t-il un vrai francophone ici qui pourrait relire ma tentative de traduction?
https://localazy.com/p/amsleser