UtilitechAS/amsreader-firmware #643: PriceAPI: blank error

joachimtingvold, 2023-09-24:
**Describe the bug**

The PriceAPI shows up in the header as if it has some error, but it's blank (i.e. contains no error message/code).

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Boot it with ENTSO API enabled
2. Immediately see the PriceAPI error in the header, where it will stay indefinitely

**Expected behavior**
No error, or at least show what the problem is.

**Screenshots**
[image]

**Hardware information:**
 - Country: Norway
 - Meter: Kaifa
 - Encryption enabled: no
 - AMS reader: Pow-U+
 - M-bus adapter (if applicable): N/A

**Relevant firmware information:**
 - Version: 2.2.21
 - MQTT: yes
 - MQTT payload type: HomeAssistant
 - HAN GPIO: N/A
 - HAN baud and parity: 2400 8E1
 - Temperature sensors: N/A
 - ENTSO-E API enabled: yes

**Additional context**
N/A

joachimtingvold, 2023-09-24:
I also see the IPv6-DNS-interpreted-as-IPv4 issue (mentioned in #563), seeing as it reports "DNS: <ipv4> / <decimal IPv6>". It is not replaced, though (compared to what was seen in #563), but that might be due to the workaround mentioned in the same issue?

Both my readers are on IPv6-enabled networks, and the decimal-encoded IPv6-address it finds as DNS is probably related to RDNSS (which I also have enabled on both these networks).

I have no idea if these two are related (the decimal-encoded IPv6 DNS and ENTSO API).

Jellybeanz1980, 2023-09-25:
I've been investigating this for some time now. I've got 2 custom-made boards running the same firmware. I've observed the issue on one of them, not on the other one. Resetting the board, even powering it down doesn't solve the issue. EntsoeAPI returns following during initiating the communication:

"(E) (EntsoeApi) Communication error, returned status: -1
(E) connection refused"

I'm not sure when the error began. Now the second board also fails. I've implemented a "ping" function in the firmware in order to check if it is a DNS issue, or a connection issue. No problem with the ping to google.com. When I copy the exact link that the board uses to connect to Entsoe, and try it in a browser of PC, it works as expected, thus no error on the API key.

gskjold, 2023-09-26:
Are you using ENTSO-E API key? Also, is this a self compiled code or official binaries?

joachimtingvold, 2023-09-26:
Not sure if you're asking me or @Jellybeanz1980, but I'll answer for my setup. Yes, using ENTSO-E API key. Using official firmware.

gskjold, 2023-09-26:
Asking all involved :) Thanks. I will try to reproduce having dns1 as ipv4 and ipv6 as dns2, maybe there is something there although I would think it should work as long as dns1 is a valid dns server

joachimtingvold, 2023-09-26:
PriceAPI used to work on both my amsreader devices. I've had DHCPv4 provided IPv4 DNS alongside RDNSS provided IPv6 DNS on my IoT VLAN since long before getting the amsreader. Not sure if it stopped working after a specific amsreader version, or if it's a combination of amsreader changes and ENTSO API changes... The ENTSO website required me to change my password when I started troubleshooting it, so I initially thought it was related to that. But even after changing the password and regenerating the API key, the result is the same.

gskjold, 2023-09-26:
Could you please try this test firmware to check if it works for you?

[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12730899/esp32s2.zip)

longer83, 2023-09-26:
Can you change primary and secondary DNS for AMS reader to 8.8.8.8 and 8.8.4.4 and check if it helped?

joachimtingvold, 2023-09-26:
I have not touched my amsreader since creating this post, and still running 2.2.21. This is the dashboard right now;

[image]
[image]

Not sure why it "suddenly" works, and why it "suddenly" stopped working... maybe it was indeed intermittent problems in the ENTSO API?

Jellybeanz1980, 2023-09-26:
In my case I use a custom ESP32 board and compile the binary file with small adjustment for my board. I've got the API key. It worked some time, and suddenly after a restart I got the error message, no matter what I do. I've tried with google DNS server IPs. I didn't change any network settings on my router, nor on the ESP32 board. Once it started failing, I can't get it work.

ArnieO, 2023-09-27:
> Not sure why it "suddenly" works, and why it "suddenly" stopped working... maybe it was indeed intermittent problems in the ENTSO API?

We have at several occasions seen issues with shorter or longer duration with the ENTSO-E API. 
That was part of the reason why we set up an intermediary server: If you activate "Enable price fetch from remote server" and leave the API key field empty, data is read from our intermediary server.

spacecosmos, 2023-09-27:
For some reason, I'm getting this today:

[image]

spacecosmos, 2023-09-27:
For some reason, I'm getting this today:

[image]

Here's the config:

[image]

Any clues to what could be the problem?

ArnieO, 2023-09-27:
> Any clues to what could be the problem?

I confirm; there is an issue. 
[Entso-E has not published data for tomorrow, 28th sep.](
https://transparency.entsoe.eu/transmission-domain/r2/dayAheadPrices/show?name=&defaultValue=false&viewType=GRAPH&areaType=BZN&atch=false&dateTime.dateTime=28.09.2023+00:00|CET|DAY&biddingZone.values=CTY|10YNO-0--------C!BZN|10YNO-2--------T&resolution.values=PT15M&resolution.values=PT30M&resolution.values=PT60M&dateTime.timezone=CET_CEST&dateTime.timezone_input=CET+(UTC+1)+/+CEST+(UTC+2))

rogerthn2019, 2023-09-27:
> ... If you activate "Enable price fetch from remote server" and leave the API key field empty, data is read from our intermediary server.
[image]
[image]

OleG-DK3460, 2023-09-28:
[image]
I have the same problem, last pricedata was sep 27th. Leaving the ENTSO-E key empty only gives "PriceAPI: Unspecified error 1". Both on PC and phone.

ArnieO, 2023-09-28:
The link in my previous post takes you to the Entso-E website, which shows that they have still not published price data for 28th September (today).
[image]

Jellybeanz1980, 2023-09-28:
Now the Entso-E website is up and running, but I'm still getting the same error message. How about you guys? Is it working for you now?

At Entsoe webpage they say "If the root CAs are different, ECP EP can reject the connection". Could it be something to do with the secure connection?

spacecosmos, 2023-09-28:
Seems to be working now. 

I had to disable the "**Enable price fetch from remote server**" setting (untick the checkbox), save settings, then enable it and save settings again. Then the correct values appeared in the dashboard. (A reboot would probably also do the trick, I would imagine.)

ArnieO, 2023-09-29:
> A reboot would probably also do the trick, I would imagine.

Yes, a reboot would have solved it after Entso-E had published the prices.

spacecosmos, 2023-09-29:
Aaaand it's blank again... This time - no error though. Anyone else?

[image]

spacecosmos, 2023-09-30:
... Back again today...

[image]

PS! Nice touch with the different colours on import vs export in the new release btw. :)

tronde-ams, 2023-09-30:
Something is weird.

Firefox 115.3.1
Generic ESP32 / Norway NO1 / own api key.

No prices in GUI.
[image]

but present at URL.
[image]

ArnieO, 2023-10-01:
> Something is weird.

A browser refresh (F5) did not fix it?

tronde-ams, 2023-10-01:
No, it did not help with a refresh. Same on another PC. Tested with Firefox and Vivaldi.
The price graph re-appeared some hours later, but with at red error warning for price Api. The prices was correct.

Just now (15:35) no price graph is shown, but both Vivaldi and FF shows prices as http://192.168.1.41/energyprice.json.

This time I have tested with one PC and one Android phone.

Seems really strange too me.

spacecosmos, 2023-10-01:
Yep - suddenly it went blank (again):

[image]

ArnieO, 2023-10-01:
Strange indeed, it could be the Entso-E server that struggles. 
One reason for setting up an intermediary server was that we had a period with a string of hickups. 
As long as our server gets a correct read-down, all our users will then be able to read prices from us. (But of course: If prices are not published, we too will be stuck.)
If you activate "Enable price fetch from remote server" without providing Entso-E API key, you will read from our intermediary server. 
What you could do next time you see such an issue is to check if Entso-E has published data, which they normally do somewhere around 13h each afternoon (for next day). [Use this URL](https://transparency.entsoe.eu/transmission-domain/r2/dayAheadPrices/show?name=&defaultValue=false&viewType=GRAPH&areaType=BZN&atch=false&dateTime.dateTime=01.10.2023+00:00|CET|DAY&biddingZone.values=CTY|10YNO-0--------C!BZN|10YNO-2--------T&resolution.values=PT15M&resolution.values=PT30M&resolution.values=PT60M&dateTime.timezone=CET_CEST&dateTime.timezone_input=CET+(UTC+1)+/+CEST+(UTC+2)), and change to the right date.

tronde-ams, 2023-10-01:
Why do the prices exist as /energyprice.json if they are not fetched from ENTSO-E?

gskjold, 2023-10-02:
Definitely something wrong in GUI here, considering that there is data in json

gskjold, 2023-10-04:
I experienced the same this morning, fix attached:
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12799451/esp32solo.zip)
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12799453/esp32.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12799454/esp8266.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12799455/esp32c3.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12799456/esp32s2.zip)

closed 2023-10-10 by gskjold

Jellybeanz1980, 2023-10-16:
Sorry for reopening this case, but I thought that an update would be in place. I've solved my issue on blank Entsoe price. I've found out that after the version 2003.07.00 of the platform-espressif32.zip the file ssl_client.cpp returns error 30592 (fatal error, connection refused) no matter which SSL site tried to be accessed. When running 2003.07.00 everything seems to be OK.

gskjold, 2023-10-17:
Thanks, did you mean 2023.08.00 vs 2023.07.00 ?

Jellybeanz1980, 2023-10-17:
Yes, exactly. I even tried later versions up to 2023.10.03, as after that the WifiClientSecure was removed. I had only luck with the version 2023.07.00