UtilitechAS/amsreader-firmware #644: No HAN data recieved

Sommerdal-ufm, 2023-09-24:
HAN data has suddenly stopped working.
Reboot doesn't fix it.

**Hardware information:**
 - Country: DK
 - Meter: 
 - Encryption enabled yes
 - AMS reader: Pow-U
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.2.21
 - MQTT: yes
 - MQTT payload type: [e.g. JSON]
 - HAN GPIO: 5
 - HAN baud and parity: [e.g. 2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]

jladefoged, 2023-09-24:
Same here and also in Denmark. I suspect it's an issue at Radius.

Sommerdal-ufm, 2023-09-24:
Thank you. I have nailed Radius

jladefoged, 2023-09-25:
I got in contact with Radius and they have requested Kamstrup to reset the HAN port. I have no idea about the timeframe, but Kamstrup will contact me by email when I can re-insert the module (it's currently removed).

Sommerdal-ufm, 2023-09-25:
I for a reply today 12:08 , that my HAN port would be reset. 12:47 I was informed by Kamstrup that the reset has been performed and it is working again.
I didn't remove it.
Problem solved.
@jladefoged Can I close the issue, or wait for you?

jladefoged, 2023-09-25:
Just close the issue. I'll wait for Kamstrup to fix mine.

Sommerdal-ufm, 2023-09-25:
Supplier problem

closed 2023-09-25 by Sommerdal-ufm

ArnieO, 2023-09-26:
> * AMS reader: Pow-U

Just to be sure future readers of the thread are not confused: 
I assume you mean Pow-K.

Unfortunately, this seems to happen with some Kamstrup meters now and then: A reset is needed, can be done remotely by the grid company. We have so far never seen this on other meter brands.

ArnieO, 2024-02-29:
We issued a blog post today with information on how the Kamstrup Omnipower issue ("No HAN data") can be remedied. 
https://www.amsleser.no/module/ets_blog/blog?id_post=33