UtilitechAS/amsreader-firmware #646: Use realtime calculated consumption when there is no data from the ams meter

stigvi, 2023-10-01:
[image]

It is a little frustrating when this happens. So my suggestion is to use the realtime calculated consumption when the value from the meter is missing for one hour. I think that would be a better solution than using zero.

stigvi, 2023-10-01:
[image]

This is how it looks like in Home Assistant where I am using power/realtime/import/hour as basis for the hourly consumption. It was 4.68kwh between 10 and 11. It would be nice if ams reader used that value instead of zero.

ArnieO, 2023-10-01:
This is not how it should work. If data is missing for on whole-hour readout, the zero should be replaced by an average at next passing of whole hour. 

What could cause this, @gskjold?

gskjold, 2023-10-02:
Not sure, but this should not happen, it should have been average of the two

gskjold, 2023-11-18:
I cannot find any good answer for this problem... And regarding using realtime calculation, I've used this before, but it causes some big issues if the calculation was too high and the meter reports a lower value later. This is a edge case I don't see the value in chasing.

closed 2023-11-18 by gskjold