UtilitechAS/amsreader-firmware #649: Future energy price missing

PerBuch54, 2023-10-03:
Showing Future energy price (undefined/100)
What does that mean ?

ArnieO, 2023-10-03:
What are your settings in this tile? 
[image]

PerBuch54, 2023-10-04:
[image]

PerBuch54, 2023-10-04:
It has been working until yesterday

gskjold, 2023-10-04:
I experienced the same this morning, fix attached:
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12799451/esp32solo.zip)
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12799453/esp32.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12799454/esp8266.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12799455/esp32c3.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12799456/esp32s2.zip)

nossaile, 2023-10-08:
I had the same issue for two days, but today it's back to normal without doing anything.
[image]

closed 2023-10-10 by gskjold