UtilitechAS/amsreader-firmware #651: Future energy price out of sync

cmock, 2023-10-06:
**Describe the bug**
In a browser tab that's been open for some days, I sometimes find the price bar graph is out of sync; reloading the page then displays the correct values.

**Screenshots**
First the bug:

[image]

After a page reload:

[image]

**Hardware information:**
 - Country: Austria
 - Meter: Sagemcom
 - Encryption enabled yes
 - AMS reader: ESP32 
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.2.22
 - MQTT: yes
 - MQTT payload type: HomeAssistant
 - ENTSO-E API enabled: yes

**Additional context**
I've only noticed this after upgrading to 2.2.22, but I also upgraded firefox to v115 at the same time.

The meter and the browser are separated by a WAN line that can at times be overloaded, I'm seeing more than 5 seconds response time for the HTTP connection during the night.

I know my way around the firefox developer tools, let me know whether I can inspect something when I next encounter the issue.

ArnieO, 2023-10-06:
I too leave it open in a browser tab - and I have seen issues like this also on previous firmware versions.
As you say: a browser page refresh (F5) fixes this (minor) inconvenience. I too run Firefox, so it may be browser related.

We need @gskjold to weigh in here, but I think this is due to optimization on how the page is updated each time new data is is pulled from the reader (each 5 second or so). Some tuning of the algorithm is maybe possible?

cmock, 2023-10-09:
Oh wow; I came back after a long weekend and now it's flickering thru various numbers of bars like crazy. The one which shows 23:00-23:00 seems to be the correct one.

Again, fixed by a reload.

https://github.com/UtilitechAS/amsreader-firmware/assets/2141141/a06e0510-d613-44e1-9be0-a29066d1fc98

gskjold, 2023-10-10:
Fancy, I think I know what this is, will fix

ArnieO, 2023-10-10:
> Oh wow; I came back after a long weekend and now it's flickering thru various numbers of bars like crazy.

I've also seen that once recently. 
Thought it was my browser that messed up, as F5 fixed it.

closed 2023-10-10 by gskjold