UtilitechAS/amsreader-firmware #655: Pow-U unable to stay connected with v2.2.22

gskjold, 2023-10-14:
We have had reports for a couple of Pow-U users that v2.2.22 is unable to stay connected to WiFi without USB power. If you have experienced this issue, please try the firmware below, and also try to disable 802.11b from config menu if problem persist:
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12903111/esp32s2.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12903112/esp8266.zip)

Changes:
- Disabled auto disconnect on low voltage
- Added option to disable 802.11b on ESP8266

JCHRA007, 2023-10-14:
Testet [esp32s2.zip]
Boots fine ... powers fine ... ;-)

[image]

ArnieO, 2023-10-14:
> Testet [esp32s2.zip] Boots fine ... powers fine ... ;-)
👍 
Running on which hardware?

JCHRA007, 2023-10-14:
Hope this helps;)

[image]

leoleonards, 2023-10-15:
Seems to work fine. Aidon.
[image]
Up 30 min.

angainor, 2023-10-15:
Thank you! This was the problem I was having also. With the wifi disconnects I could not understand what's going on. The device was working, blinking red, did not re-connect. I did have 'reboot on connection error' checked, but I guess this case did not fall into that category? after all, there was no connection error, wifi was turned off on purpose?

FWIW, I think it's best to let the device happily reboot when it runs out of power. It only takes a few seconds, you won't loose much data, the device is back quickly. I never noticed any problems until now..

ArnieO, 2023-10-16:
> Thank you! This was the problem I was having also. With the wifi disconnects I could not understand what's going on. The device was working, blinking red, did not re-connect. I did have 'reboot on [...]

If it happens often, and especially if you are on a meter configured according to HAN-NVE standard, you might loose (sometimes of often) the whole hour payload - which contains Energy  reading (kWh). 

HAN-NVE standard is used in
- Norway (all meters) 
- Parts of Sweden (grid companies that rolled out automatic meters before Sweden decided to go for the P1 standard)

gskjold, 2023-10-16:
> Thank you! This was the problem I was having also

Good to hear it was resolved

> If it happens often, and especially if you are on a meter configured according to HAN-NVE standard, you might loose (sometimes of often) the whole hour payload - which contains Energy reading (kWh).

And also, realtime hourly data gets reset reset if it reboots from brownout or hardware reset, which could be devestating if you run any sort of home automation rules around that. Only software reboot preserves realtime data.

Releasing 2.2.23 soon

closed 2023-10-16 by gskjold

angainor, 2023-10-16:
@ArnieO @gskjold OK, I understand. But in my case I am not concerned, because I integrate all data in HomeAssistant, and I don't rely on Pow-U to collect it. I only depend on Pow-U to deliver the momentary power usage values regularly. HomeAssistant computes the hourly usage, cost, etc.

Of course others might need the Pow-U data collection, but maybe then this behavior could be configurable? In my case Pow-U simply stopped delivering any data to HA for many hours, and I didn't have any idea why. That was quite devastating for my home automations ;)

ArnieO, 2023-10-16:
> My esp32 stopped working on the v2.2.22, hooked it up to the computer and it seemed fine. update to v2.2.23 and it seems to behave poorly still. changed the wifi settings to power savings off, still not working as it used to.

- Which hardware are you running?
- Did the observed change come from v2.2.21 to v2.2.22?
- It would be helpful if you can be more specific regarding "stopped working", "behave poorly" and "not working as it used to"?

kng, 2023-10-16:
It's a Olimex ESP32-DevKit-LiPo on a custom interface. It has been working for quite a long time now.
Updated from v2.2.21 to v2.2.22 a few days ago and it immediately lost connectivity and never came back. power cycled etc, didn't have time to investigate so left it alone for a few days.
Hooked it up to the computer and it came online, so fw upgrade seems to have went ok.
I put it back on the power meter and it came online for a few hours and then it disconnected again.
Updated to v2.2.23 tonight and it behaved pretty much the same, changed the wifi settings a bit but no change.
Have now downgraded to v2.2.21 and will see how it does.

ArnieO, 2023-10-16:
@kng Thank you for additional information! 👍

kng, 2023-10-17:
It has been running fine all night. Could it be related to the [handleVoltageCheck()](https://github.com/UtilitechAS/amsreader-firmware/blob/master/src/AmsToMqttBridge.cpp#L847-L865) ? Because when running on USB it seems fine. But when running on the regulated 3.3V it acts up somehow.

gskjold, 2023-10-17:
Strange. In 2.2.23, the voltageCheck has been disabled.

kng, 2024-05-26:
This still seems to be an issue, tried latest v2.3.5 and not working when connected to meter.
Messed around with the voltage measurement, also disabled it (with setting it to 0).
I just have to stay on v2.2.21 for it to work.

ArnieO, 2024-05-27:
The current demand from the ESP32 varies a lot, and this application exploits the wi-fi modem heavily. To prevent the ESP32 from going into brownout and potentially reboot it is highly recommended to have a quite large low-ESR capacitor on the 3,3V power rail. 

There are a number of closed threads here where users running their own hardware realize their issues are related to unstable 3,3V power.