UtilitechAS/amsreader-firmware #657: Real time calculation shows wrong values.

OMJOB, 2023-10-16:
**Describe the bug**

Export Day shows 69.5kWh (between midnight and 11:30) which is not possible as the inverter cannot produce that much, according to the inverter I have produced 9kWh (between 08:00 and 11:30) which is more reasonable.
Import Day shows 39.8kWh but can't have used more than max 4kWh since midnight.

**To Reproduce**
Steps to reproduce the behavior:
1. No idea.

**Expected behavior**
Show correct values.

**Screenshots**
[image]

**Hardware information:**
 - Country: Sweden
 - Meter: Kamstrup
 - Encryption enabled no
 - AMS reader: Device: Pow-K+ Chip: esp32s2 (160MHz)

**Relevant firmware information:**
 - Version: v2.2.23
 - MQTT: yes
 - MQTT payload type: HomeAssistant
 - HAN GPIO:
        gpioHanPin 16
        gpioHanPinPullup 1
        gpioApPin 0
        gpioLedPinRed 13
        gpioLedPinGreen 14
        gpioLedRgbInverted 1
        gpioVccPin 10
 - HAN baud and parity: 2400 8N1
 - Temperature sensors Don't know
 - ENTSO-E API enabled: yes

gskjold, 2023-10-16:
In your config, enable multipliers under meter config and set 0.1 kwh. Graph data from earlier is invalid and will affect your current realtime values. To get correct values from now on, you can upload attached config file to reset everything to zero: 
[reset-data.cfg.txt](https://github.com/UtilitechAS/amsreader-firmware/files/12915460/reset-data.cfg.txt)

Rename from .cfg.txt to .cfg before uploading

OMJOB, 2023-10-17:
This seems to have solved the problem, a big thanks for the good and quick help.

closed 2023-10-17 by OMJOB

ArnieO, 2023-10-17:
> This seems to have solved the problem, a big thanks for the good and quick help.

For some reason, we have seen that there are some (few and individual) energy meters that deliver a different number of decimals compared to what is common. We have no idea why this problem exists.

The content of the payloads in those "deviant" meters do not include enough information to place the decimal point at the right position. The way to get correct readings from those meters is to make a correction using the multipliers, as explained by @gskjold here above.