UtilitechAS/amsreader-firmware #658: PriceAPI: Internal server error after upgrade to V2.2.23

eireit, 2023-10-17:
After firmware update from v2.2.22 to the new 2.2.23 i have recieved the fault:  PriceAPI: Internal server error
Have tested new token from Enstoe. still smae error.

Device information
Chip: esp32s2 (160MHz)
Device: [Pow-U+]

Meter
Manufacturer: Kaifa
Model: MA304H3E

Mvh Eirik

ArnieO, 2023-10-17:
There are occasional hickups with the price reading from Entso-E API. If you just remove the Entso-E token, prices are read from our intermediate server, which is usually more reliable (providing Entso-E has published prices at all).

This is the recommended setting:
[image]

Allan-Hoejberg, 2023-10-17:
I have the same issue after upgrading to 2.2.23
/Allan

gskjold, 2023-10-18:
Regarding running with API key, ENTSO-E is currently sending an error with message "Unexpected exception, please contact support and provide exception ID", I am not sure what their problem is.

Using the service without API key will pass though our servers which is most likely affected by the same problem for many price zones. I will try to get an overview.

tb205gti, 2023-10-18:
> Regarding running with API key, ENTSO-E is currently sending an error with message "Unexpected exception, please contact support and provide exception ID", I am not sure what their problem is. [...]

I have created a support ticket at entsoe for this issue.

tb205gti, 2023-10-18:
Seems it is working again

gskjold, 2023-10-18:
Indeed, it seems to work now. Painfully slow at times though

tb205gti, 2023-10-18:
Got validation from entsoe - it was an error on their end. Hotfix have been deployed

closed 2023-10-18 by ArnieO