UtilitechAS/amsreader-firmware #664: Changing from CEST to CET

mortenm22, 2023-10-28:
**Describe the bug**
One hour is missing in the json price

**Screenshots**
[image]

**Hardware information:**
 - Country: [e.g. Norway]
 - Meter: [e.g. Kamstrup]
 - Encryption enabled [no]
 - AMS reader: [Pow-K+]
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: [v2.2.23]
 - MQTT: [yes]
 - MQTT payload type: [JSON]
 - ENTSO-E API enabled: [yes]

**Additional context**
It looks like it's the last price missing. We have two prices for 2 o'clock this day which also look like they have been swapped.

ArnieO, 2023-10-28:
It's daylight saving time night again...! 😅

gskjold, 2023-10-28:
I'll soon start a rebellion against this "daylight saving" bs...

gskjold, 2023-10-28:
This turns out to be a two-part issue, one being display in user interface, the other is a issue in our intermediate server where the first 02:00 price is overwritten by the second. I fixed the first one on my local branch, but the latter issue was a bit deeper than I first thought. I will continue in the morning

gskjold, 2023-10-29:
This issue have been fixed. If you are missing price for 23.00 today, please reboot your device to make it re-fetch new prices.

GUI changes have been committed  for next patch, ready for next year

mortenm22, 2023-10-29:
I have all prices now. Thank you so much!

mortenm22, 2023-10-29:
After reboot of Pow-K+ I have all prices in json string but not in user interface. ENTSO-E API enabled: [no]
[image]
CET 29.10.2023 kl 11.00
{
  "id": "D4:F9:8D:6E:87:10",
  "prices": {
    "0": 0.5009,
    "1": 0.4946,
    "2": 0.4937,
    "3": 0.5037,
    "4": 0.6506,
    "5": 0.7175,
    "6": 0.7786,
    "7": 0.8049,
    "8": 0.7738,
    "9": 0.7189,
    "10": 0.6594,
    "11": 0.6159,
    "12": 0.4854,
    "13": -127,
    "14": -127,
    "15": -127,
    "16": -127,
    "17": -127,
    "18": -127,
    "19": -127,
    "20": -127,
    "21": -127,
    "22": -127,
    "23": -127,
    "24": -127,
    "25": -127,
    "26": -127,
    "27": -127,
    "28": -127,
    "29": -127,
    "30": -127,
    "31": -127,
    "32": -127,
    "33": -127,
    "34": -127,
    "35": -127,
    "36": -127,
    "37": -127,
    "min": 0.4854,
    "max": 0.8049,
    "cheapest1hr": "2023-10-29T22:00:00Z",
    "cheapest3hr": "2023-10-29T10:00:00Z",
    "cheapest6hr": "2023-10-29T10:00:00Z"
  }
}

mortenm22, 2023-10-29:
one more reboot and the display is correct too.
[image]

mortenm22, 2023-10-29:
Thank you very much for a very good job.

mortenm22, 2023-10-29:
After prices for tomorrow came, I'm missing price for hour 23. I've tried rebooting. Price for hour 23 is still missing.
{
  "id": "XX:XX:XX:XX:XX:XX",
  "prices": {
    "0": 0.7787,
    "1": 0.8049,
    "2": 0.7738,
    "3": 0.7189,
    "4": 0.6594,
    "5": 0.6159,
    "6": 0.4854,
    "7": 0.2195,
    "8": 0.1488,
    "9": 0.1402,
    "10": 0.1367,
    "11": 0.2393,
    "12": 0.577,
    "13": 0.7015,
    "14": 0.9125,
    "15": 0.9695,
    "16": 0.9138,
    "17": 0.8757,
    "18": 0.8554,
    "19": 0.8402,
    "20": 0.8614,
    "21": 0.9598,
    "22": 1.0498,
    "23": 1.1273,
    "24": 1.2783,
    "25": 1.1918,
    "26": 1.0235,
    "27": 0.9264,
    "28": 0.913,
    "29": 0.9378,
    "30": -127,
    "31": -127,
    "32": -127,
    "33": -127,
    "34": -127,
    "35": -127,
    "36": -127,
    "37": -127,
    "min": 0.1367,
    "max": 1.2783,
    "cheapest1hr": "2023-10-30T02:00:00Z",
    "cheapest3hr": "2023-10-30T00:00:00Z",
    "cheapest6hr": "2023-10-29T22:00:00Z"
  }
}

mortenm22, 2023-11-01:
For the last day, the prices have been correct. I am therefore closing the case.

closed 2023-11-01 by mortenm22