UtilitechAS/amsreader-firmware #666: upgrade not possible

arkimos, 2023-10-31:
OTA upgrade  is not possible. Also tryed to upload .bin file. 
[image]

**Hardware information:**
 - Country: Switzerland
 - Meter: Landis & Gyr E450
 - Encryption enabled: no
 - AMS reader: esp32c3
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version:  v2.2.21
 - MQTT: no 
 - MQTT payload type: [e.g. JSON]
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [e.g. 2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]

ArnieO, 2023-10-31:
We see this from time to time, and are not sure what the reason is. Could be some setting in the Wi-Fi router that prevents, or could be latency / timeout during the download process. 

Since you are on ESP32c3 I assume this is your own hardware. And then there is a third potential issue: Power consumption. The devices consume more power while downloading compared to normal operation, as there is continuous Wi-Fi activity over a relatively long time. Is your voltage sufficiently stable?

There two other ways to upgrade: 
- USB port
- UART

In both cases, you can follow the guideline in the Firmware User Manual, chapter "9 Flashing the device over USB cable".

arkimos, 2023-10-31:
Solved.
Power consumption was the Problem! 
Tryed with stable power supply and it worked.. 
Tanks for your support.

closed 2023-10-31 by arkimos