UtilitechAS/amsreader-firmware #667: Increase MQTT header buffer size

bbergshaven, 2023-11-01:
We have run into a problem with too long MQTT passwords that makes it impossible to connect the device to our MQTT server. The problem is described in the following issue: https://github.com/256dpi/arduino-mqtt/issues/296
We are using JWTs as password for our integrations.

We are kindly asking for:
a) Increasing the buffer size to 256
b) Allow to set the buffer size manually in the MQTT config

The line that needs to change is:
https://github.com/UtilitechAS/amsreader-firmware/blob/206c90ca468c4ff8234f12d56070e913bdb29567/src/AmsToMqttBridge.cpp#L1812C14-L1812C25

We are currently working with a customer that is planning to buy a large number of units from amsleser.no, but needs this to be able to use the unit.

gskjold, 2023-11-01:
A quick fix with 256 buffer size attached:
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13231678/esp32s2.zip)
If it works, we'll also add the configuration. Maybe this should also self-adjust based on credential size. May I ask the actual size of the jwt used? Number of characters

bbergshaven, 2023-11-01:
Fantastic response time! 🙏
Will test tomorrow and let you know!

The size of the jwt used is 104 characters.

bbergshaven, 2023-11-03:
Hello @gskjold 
Should it work to update the firmware over wifi (using the firmware.bin) and if so, will the version string in the UI update?

gskjold, 2023-11-03:
Yes, but if you are using pow-u or similar BUS powered device, you will need to connect USB for extra power. What hardware do you use? The version will update in the GUI.

And by the way, unzip the zip file and upload "firmware.bin"

bbergshaven, 2023-11-03:
This is the hw: https://amsleser.no/hjem/7-5-pow-u-universal.html#/5-pow_u_options-pow_u_med_boks_og_15_cm_rj_45_kabel

Tried to do an update, but after the reboot the version string shows the same as before, what should be the version string in the GUI after update?

gskjold, 2023-11-03:
OK, you need USB power while running the upgrade at least. Version should be 63c3d31

bbergshaven, 2023-11-06:
Managed to update the unit and the fix worked perfectly and we now have streaming data to our MQTT broker. Thank you 🥇

gskjold, 2023-11-08:
Excellent news! I might just keep the 256 buffer size as default. It will be included in next patch. If I have not released that before your client wants to order, let me know and I'll release

closed 2023-12-04 by gskjold