UtilitechAS/amsreader-firmware #674: Prices loaded are from yesterday - fixes when restarting the unit

KennethLavrsen, 2023-11-15:
**Describe the bug**
I have seen days in a row that the prices loaded are wrong and it is clear that the prices shown are from the day before

**To Reproduce**
Check the prices in the morning or mid day and compare with a known good source and also compare with the day before.
The price may be wrong (not always but often enough to notice).
A restart of the AMS reader forces it to reload prices and then it has always fixed itself

**Expected behavior**
Prices expected to be correct

**Screenshots**
N/A

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup
 - Encryption enabled yes
 - AMS reader: Pow-K
 - M-bus adapter (if applicable): N

**Relevant firmware information:**
 - Version: 2.2.23
 - MQTT: yes
 - MQTT payload type: Home assistant
 - HAN GPIO: 
 - HAN baud and parity: 
 - Temperature sensors [
 - ENTSO-E API enabled: yes

**Additional context**

I have seen the exact same thing happen with the Energi Data Service integration for Home Assistant. It seems the source of the data often post the update later than they should. The problem for the EDS was resolved by hacking her code to load data one hour later.

gskjold, 2023-11-15:
With or withour ENTSO-E API key?

KennethLavrsen, 2023-11-15:
I have my own key

gskjold, 2023-11-16:
Testing some adjustments:
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13381137/esp32solo.zip)
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13381138/esp32.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13381140/esp8266.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13381141/esp32c3.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13381142/esp32s2.zip)

KennethLavrsen, 2023-11-16:
Thanks. I did not see the error this morning. But I installed the dev version you posted above anyway and will report if I see anything good or bad over the weekend