UtilitechAS/amsreader-firmware #677: There might be a problem also in the test version "d82e705"

LA5HJA, 2023-11-20:
**Describe the bug**
I'm running both version 2.2.23 and d82e705 and today they both did show price in "øre" and low numbers like the problem with € to NOK.

Another problem i did find is that my peak is at 13.79kWh (on both versions) yesterday but non of the last 24 hour is over 7,9kWh.

**To Reproduce**
Not sure how to do reproduce the error.

**Screenshots**
[image]
[image]

**Hardware information:**
 - Country:  Norway
 - Meter: Aidon
 - Encryption enabled no
 - AMS reader: ESP32-devkit1
 - M-bus adapter KINA!

gskjold, 2023-11-20:
Interesting. But also, what browser and screen resolution is this?

gskjold, 2023-11-20:
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13415791/esp32solo.zip)
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13415792/esp32.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13415793/esp32s2.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13415794/esp8266.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13415795/esp32c3.zip)