UtilitechAS/amsreader-firmware #678: License change

dbeinder, 2023-11-20:
Relicensing away from GPL generally requires consent from all contributors. Of course, since there aren't many external contributions, it shouldn't be to difficult to 'reimplement' the relevant commits (in whatever way that is defined in a legal sense).

I understand your motivation, but if you had been planning to do this, you should not have accepted pull requests without an explicit CLA.

If this is in response to gplug.ch selling devices with your firmware, I'd like to note I am not affiliated with them, but they may have based their HW on my design - I haven't purchased one of their device to verify that.

gskjold, 2023-11-21:
I have to admit that I have been too focused on other aspects of the legal side of things to cover this properly. But I do understand what you are saying here and I would like to propose two options to remedy this:

1. Invite all contributors to voice their opinion in this thread

or

2. Start a new branch (main) on the commit before this, setting this to the new primary branch. Then start a new branch (license_change) for this commit to a new branch and create a pull from there and invite all contributors to review.

I'm not sure which one (if any) solves this misstep in the best way. I am open for input on this one. As you say there are not many external contributors during our period on GPL (five external), and the contributions are mostly smaller things.

gskjold, 2023-11-25:
Reverted and created PR https://github.com/UtilitechAS/amsreader-firmware/pull/682

closed 2023-11-25 by gskjold