UtilitechAS/amsreader-firmware #681: Pow-K+ regularly reboots with "Software reset (0/3)"

ArnieO, 2023-11-23:
**Describe the bug**
Customer reports of a Pow-K+ that reboots regularly with "Software reset (0/3)", like this:
[image]
"Auto reboot on connection problem" is deactivated.

We have currently not been able to find the reason for this, and there are very few points in the code that could cause Software reset. 

We open this Issue to track progress. 

**Expected behavior**
This is not expected behaviour

**Hardware information:**
 - Country: Norway
 - Meter: Kamstrup
 - Encryption enabled: No
 - AMS reader: Pow-K+

**Relevant firmware information:**
 - Version: 2.2.23
 - MQTT: Yes
 - MQTT payload type: Raw
 - HAN GPIO: 16
 - HAN baud and parity:2400 8N1
 - Temperature sensors: No
 - ENTSO-E API enabled: Yes

gskjold, 2023-11-25:
Removing software reboot on connection error option, this have only caused issues for us anyway...

Firmware:
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13463513/esp32.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13463514/esp8266.zip)
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13463515/esp32solo.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13463516/esp32c3.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13463517/esp32s2.zip)

Other places software reboot happens:
- When initial setup is saved
- When disabling debugging
- When WiFi config is changed
- Call via browser to reboot
- Factory reset
- After firmware upgrade
- After uploading configuration file

ArnieO, 2023-11-25:
Thank you for the listing, it is then puzzling how software reboot can be triggered during normal operation. 
Good idea to try removing "Software reboot on connection error", it will be interesting to hear if it helps!

rydrik, 2023-11-26:
Device reboots even with Auto boot off:
[image]

ArnieO, 2023-11-26:
> Device reboots even with Auto reboot off:

Can you try the test version here? https://github.com/UtilitechAS/amsreader-firmware/issues/681#issuecomment-1826264748

sandjo, 2023-11-26:
My Pow-P1 reboots regularly, I think almost every night. Since it doesn't really affect anything I haven't kept track. But I saw this issue and saw the reboot time. From my device information:

Last boot: 26.11.2023 05:23
Reason: Software reset (3/0)

Just a coincidence that they rebooted at the same time, or is there anything special happening at that time?

I have auto reboot disabled in my configuration.

rydrik, 2023-11-26:
Have upgraded to 0e6abd0 now, will keep you updated.

ArnieO, 2023-11-26:
> Just a coincidence that they rebooted at the same time, or is there anything special happening at that time?

Potentially interesting! We are chasing a hard-to-find bug here, so all such observations are welcome. 
But there should be nothing in the code triggering something at a specific time - and "Software reset" should be triggered by the code (unless the ESP for some reason reports incorrectly).
Could you too be so kind to install the test version here above, and report back?

sandjo, 2023-11-27:
Sorry if I gave the impression that (in my case) it is always happening around that time I mentioned. It is always during the night, but not at any fixed time. Todays restart was earlier and I believe at a more usual time. Since I only have anecdotal evidence I don't want to sound too certain, but the times I've checked the time in the web interface it has usually been around 02-ish when the reset happened.

Chip: esp32s2 (160MHz)
Device: Pow-P1
Last boot: 27.11.2023 01:49
Reason: Software reset (3/0)

This Pow-P1 is installed at our vacation house and I won't have physical access to it (for recovery if needed) before Christmas, so I'm a bit reluctant to install test versions until then. The daily resets are not currently causing me any problems either.

More info about my installation just in case it rings any bells
 - auto reboot disabled as noted before
 - legacy 802.11b disabled
 - quiet network at the moment and good signal quality (-46 dBm)
 - only used via the web interface sporadically by a human
 - prices fetched from ENTSO-E with API key, but I often get the wrong currency displayed for "Future energy price" (øre instead of SEK)

ArnieO, 2023-11-27:
@sandjo 
> I'm a bit reluctant to install test versions until then

No worries, but thank you for your update anyway!

@toast1234 
> I have 2 ESP32 devices in different location connected to different meters. They reboot every night at the same time.

Very interesting for two reasons:
1) Same time two units and each night
2) I have never before seen the "Software reset CPU (12/0)" message reported. 

Thank you for contributing with this information!
However, I think this could be something entirely different. Both because it is a different message, and maybe because this apparently is your own hardware. But indeed interesting that "something" happens at the same time.

ArnieO, 2023-11-27:
> It stayed up ~21h with test firmware, rebooted with (1/0) instead of (3/0), if it makes any difference:

Indeed, this is important information. Thank you!
Have you deactivated "Allow 802.11b legacy rates"?

This is the recommended setting, as 802.11b legacy rates is only needed for very old routers:
[image]

rydrik, 2023-11-27:
Yes, "Allow 802.11b legacy rates" is disabled. With v2.2.23, the Pow-K+ rebooted with both **Vbat power on reset (3/0)** and **Vbat power on reset (1/0)**, but with test version 0e6abd0 reboot cause is **Vbat power on reset (1/0)** so far. I will update with further reboot info :)

ArnieO, 2023-11-27:
> Yes, "Allow 802.11b legacy rates" is disabled. With v2.2.23, the Pow-K+ rebooted with both **Vbat power on reset (3/0)** and **Vbat power on reset (1/0)**, but with test version [...]

We have investigated quite thoroughly those occasional power reboots "Vbat power on reset (1/0)" on Pow-K, including a technical discussion with Kamstrup on the power delivery to the 6-pin plug in the expansion bay. 

Our impression is that most users got rid of that problem after disabling "Allow 802.11b legacy rates". On my Kamstrup the problem stopped completely. 

For the few of you that still see it: It is **maybe** caused by the power meter prioritising power to other parts of it so (maybe coinciding with the readout by radio to the grid company), causing the voltage to drop to a level where a voltage supervisor on the board reboots the device to avoid brownout. That is something we will most probably have to live with. but on next generation layout we will lower the detection level for that voltage supervisor a bit - giving us a bit more margin hence reduced risk of such hardware reboots.

rydrik, 2023-11-27:
> Yes, "Allow 802.11b legacy rates" is disabled. With v2.2.23, the Pow-K+ rebooted with both **Vbat power on reset (3/0)** and **Software reset (1/0)**, but with test version [0e6abd0](https://github.com/UtilitechAS/amsreader-firmware/commit/0e6abd0f186c721c655d599405df3117264918e0) reboot cause is **Vbat power on reset (1/0)** so far. I will update with further reboot info :)

Copy paste error in last post corrected...

gskjold, 2023-11-28:
Very interesting indeed. Could this be wifi related somehow? I know ubiquiti unifi can run a channel optimization during the night if enabled

rydrik, 2023-11-28:
Have ASUS AiMesh with a total of four nodes, but the Pow-K+ doesn't seem to change connection. Rebooted again 03:24 with reason "Vbat power on reset (1/0)". I see there is a debug possibility, problem is that I don't know when the reboot will occur. Would be good to know why it reboots instead of just guessing.

ArnieO, 2023-11-28:
> I don't know when the reboot will occur. Would be good to know why it reboots instead of just guessing.

I put a voltage logger on my Pow-K when I had this issue, measuring the internal Vcc (3,3V) and found that the "Vbat power on reset" was a result of voltage drops that lasted for up to around 10 seconds, sometimes (not every time) draining the supercap enough to cause a hardware reset. Disabling "Allow 802.11b legacy rates" fixed it for me - so a fair conclusion is that it was a result of over consumption of power on the Pow-K linked to abnormal/unexpected WiFi activity. 

The feedback we have received indicates that disabling "Allow 802.11b legacy rates" has resolved this issue for most users - but apparently not all. My hypothesis is that there is something going on in the WiFi network that somehow keeps the ESP32 busy in Wi-Fi for long enough to pull down the voltage and cause a hardware reset.

rydrik, 2023-11-29:
Upgraded FW in one of the AiMesh devices, then the Pow-K+ rebooted (think it was **Vbat power on reset 3/0**). Then it was up 23h before I went to work this morning, not sure if it has rebooted more but it did with **Vbat power on reset (1/0)** 20min ago. If it was related to my WiFi, shouldn't the reboots be consistent? It varies between several times per day to a few days inbetween, it seems.

ArnieO, 2023-11-29:
> If it was related to my WiFi, shouldn't the reboots be consistent?

In which way consistent? If it is somehow linked to your Wifi, I assume it could depend on what else is going on in your network. 

That said, there could be other reasons.  During our technical dialogue with Kamstrup, they confirmed there is a feature in the meter that prioritizes power to the parts of the meter that needs it at any time - and the HAN port has low priority. They confirmed that there is a theoretical possibility that power delivery to the HAN port could suffer at the time where the meter communicates on the radio network that collects data from the meter. If for the distance to your neighborhood node is long, RF power on that transmission may have to be increased - and the result could be that the HAN port is current limited, causing voltage to drop enough to trigger a HW reset in the Pow-K, which gives "Vbat power on reset (1/0)".

However, I see no practical way to confirm or discard that hypothesis. This is genuinely tricky!

rydrik, 2023-11-30:
Ok, so this will probably not be resolved until you release a new version of the Pow-K+ ?

ArnieO, 2023-11-30:
> Ok, so this will probably not be resolved until you release a new version of the Pow-K+ ?

Next version Pow-K will have more margin, as we will lower the hardware reboot level from 3,08V to 2,93V (nominal threshold on the voltage supervisor chip). 

But as we do not fully understand the mechanism, all we can promise is that the new version will be more tolerant to eventual short current delivery drops from the Kamstrup.

rydrik, 2023-11-30:
Anything else you would like me to test? If not, guess I can revert to v2.2.23?

rydrik, 2023-11-30:
> > Anything else you would like me to test? If not, guess I can revert to v2.2.23?
> 
> No, not really. My understanding is that "Software reset (0/3)" disappeared with test version [0e6abd0](https://github.com/UtilitechAS/amsreader-firmware/commit/0e6abd0f186c721c655d599405df3117264918e0), but you still/now have "Vbat power on reset (1/0)".
> 
> This seems to confirm that the modification in [0e6abd0](https://github.com/UtilitechAS/amsreader-firmware/commit/0e6abd0f186c721c655d599405df3117264918e0) made a difference - which is important information.
> 
> We might release some new test code to further investigate "Software reset (0/3)", which is the main subject of this thread.

I can leave the [0e6abd0](https://github.com/UtilitechAS/amsreader-firmware/commit/0e6abd0f186c721c655d599405df3117264918e0) for some days if you want to make sure I only get **Vbat power on reset (1/0)** reboots with it? And when I revert to v2.2.23, I should only get **Software reset (0/3)** reboots?

ArnieO, 2023-11-30:
> I can leave the [0e6abd0](https://github.com/UtilitechAS/amsreader-firmware/commit/0e6abd0f186c721c655d599405df3117264918e0) for some days if you want to make sure I only get **Vbat power on reset [...]

Yes, that is what we expect - and it would be great if you could test it as you indicate!

ArnieO, 2023-11-30:
We have an indication that there might be a link between this issue and https://github.com/UtilitechAS/amsreader-firmware/issues/665
In that case, the issue is that the Norges Bank API has stopped responding during nighttime.

To those of you helping us with testing, using Entso-E API key:
Can you try with latest firmware release (v2.2.23):
- One night with Entso-E API key
- One night without Entso-E API key
 
Leave it overnight each time, and see if there is a difference. 

If the hypothesis holds, there could be a nightly reboot "Software reset (0/3)" with API key, but not if you run it without API key (it reads prices from our intermediate server).

rydrik, 2023-12-02:
> Yes, that is what we expect - and it would be great if you could test it as you indicate!

I am still on [0e6abd0](https://github.com/UtilitechAS/amsreader-firmware/commit/0e6abd0f186c721c655d599405df3117264918e0) but got a **Software reset (3/0)** reboot 1900 yesterday, up since 19h now.

ArnieO, 2023-12-02:
> > I have also these kind of reboots (Software reset (0/3) frequently. They also mostly occur after midnight, before morning. ( 00.00 - 07.00). As recommended above I just disabled ""Allow 802.11b legacy rates" "
> 
> Are you seeing any changes?
> 
> No I still have the daily reboot. These are the three latest reboots: Last boot: 29.11.2023 01:45 Reason: Software reset (3/0)
> 
> Last boot: 30.11.2023 01:49 Reason: Software reset (3/0) Last boot: 01.12.2023 01:06 Reason: Software reset (3/0)

Are you using Entso-E API key for reading prices?

toast1234, 2023-12-02:
> To those of you helping us with testing, using Entso-E API key: Can you try with latest firmware release (v2.2.23):
> 
> * One night with Entso-E API key
> * One night without Entso-E API key

I know that I don't have the same hardware or reset code. But I thought that I would anyway report as the result might be relevant to perhaps tick of one issue. As you know my device reboot only once every night so I have now disable Entso on one device and it has not rebooted since. I rebooted once maualy after removing the key as the GUI reported price API failed -1 after just saving the settings.
`date && curl -s http://192.168.16.110/data.json | jq .u/60/60 && curl -s http://192.168.5.110/data.json | jq .u/60/60
Sat Dec  2 16:05:44 CET 2023
15.020833333333334
48.86583333333333`

Nealux, 2023-12-02:
> I have also these kind of reboots (Software reset (0/3) frequently. They also mostly occur after midnight, before morning. ( 00.00 - 07.00). As recommended above I just disabled ""Allow 802.11b legacy rates" "

Are you seeing any changes?

No I still have the daily reboot. These are the three latest reboots: Last boot: 29.11.2023 01:45 Reason: Software reset (3/0)

Last boot: 30.11.2023 01:49 Reason: Software reset (3/0) Last boot: 01.12.2023 01:06 Reason: Software reset (3/0)

Are you using Entso-E API key for reading prices?

Yes I have "Enable price fetch from remote server" enabled with a Entso key.
The device was rebooted for the forth consecutive night last night:
Last boot: 02.12.2023 01:08
Reason: Software reset (3/0)

I will disable the fetching of price to see what that will do.

rydrik, 2023-12-02:
> Huh... interesting. And the fact that it happened at 19h kind of undermines my hypothesis in [#681 (comment)](https://github.com/UtilitechAS/amsreader-firmware/issues/681#issuecomment-1833844596).
> Are you using Entso-E API key for reading prices?

Yes, I use the Entso-E API key.

Nealux, 2023-12-03:
> Are you seeing any changes?
>
> No I still have the daily reboot. These are the three latest reboots: Last boot: 29.11.2023 01:45 Reason: Software reset (3/0)
>
> Last boot: 30.11.2023 01:49 Reason: [...]

Just after the price update was disabled last afternoon the device rebooted but no reboot during the night.
I will keep the price update disabled for a while if it is of any interest.

ArnieO, 2023-12-04:
@sandjo Thank you for very useful feedback!

rydrik, 2023-12-04:
> > Huh... interesting. And the fact that it happened at 19h kind of undermines my hypothesis in [#681 (comment)](https://github.com/UtilitechAS/amsreader-firmware/issues/681#issuecomment-1833844596).
> > Are you using Entso-E API key for reading prices?
> 
> Yes, I use the Entso-E API key.

Do you want me to try to disable the key while on https://github.com/UtilitechAS/amsreader-firmware/commit/0e6abd0f186c721c655d599405df3117264918e0 or do you want me to go back to v2.2.23 and test this? The meter reader have not rebooted in over 2 days now.

Nealux, 2023-12-04:
> Just after the price update was disabled last afternoon the device rebooted but no reboot during the night.
I will keep the price update disabled for a while if it is of any interest.

Can you try to enable price fetch, but without the Entso-E API? In that way, prices are fetched from out intermediate server. It will help figuring out what could cause the reboot.

Now two nights without reboot.
That is without  any price fetch.
Today I enabled the price fetch but without Entso-E- API key.
Let's see how this will go.

rydrik, 2023-12-09:
I have had v2.2.24 running for over 3 days now witout any reboots, price fetch with Entso-E API key.

sandjo, 2023-12-11:
Another data point: I updated to 2.2.24 on 04.12.2023 and have been running with ENTSO-E API key enabled without resets since then.

rydrik, 2023-12-13:
> I have had v2.2.24 running for over 3 days now witout any reboots, price fetch with Entso-E API key.

Got a **Software reset (3/0)** 23:20 yesterday, then the reader had been up 6-7 days. Can't remember last time it was up this long :)

gskjold, 2023-12-21:
Thanks for all your feedback. Considering this resolved

closed 2023-12-21 by gskjold