UtilitechAS/amsreader-firmware #683: HAN RX errors (unsupported frames?)

mikkel75, 2023-11-30:
I have had a very stable system running for a long time - no RX errors.
The last week or so I have noticed a lot of RX errors starting to appear.

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup Omnipower
 - Encryption enabled yes
 - AMS reader: ESP8266
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: v2.2.23
 - MQTT: yes
 - MQTT payload type: JSON
 - HAN GPIO: UART2
 - HAN baud and parity: 2400 8E1
 - Temperature sensors NO
 - ENTSO-E API enabled: no

**Additional context**
I have attached some debug logs to maybe give a hint to the issue. 
Could it be new frames from Kamstrup that are not supported?

[HAN.txt](https://github.com/UtilitechAS/amsreader-firmware/files/13509775/HAN.txt)
[HAN_2.txt](https://github.com/UtilitechAS/amsreader-firmware/files/13509777/HAN_2.txt)

ArnieO, 2023-11-30:
The firmware is very flexible wrt frame content. As long as the standard is followed, data will be accepted and interpreted. And I find it very unlikely that there is a change on your power meter.

All valid frames start and end with "7E". I see some of your captured frames are not starting correctly, so this looks like a data read error in the hardware. 

What hardware is this? (Wild guess: Wemos D1 mini clone)
Most problems with ESPs tend to boil down to unstable power supply, which often means inadequate decoupling, as the current pulses drawn during WiFi transmission are quite large. Most Wemos D1 mini clones have very little decoupling. Try adding a huge ceramic or tantalum capacitor on 3,3V.

mikkel75, 2023-12-03:
My setup has been running reliably for a long time, but after update to 2.2.23 (was running an old version) I start to see this RX errors.

BTW The ESP8266 is heavily decoupled, so not likely a power issue.

gskjold, 2023-12-04:
I would also try to re-seat whatever connection you have to your meter. There have been a few cases where that fixes similar problem with Kamstrup

mikkel75, 2023-12-04:
Seems to be a low-voltage problem.

closed 2023-12-04 by mikkel75

ArnieO, 2023-12-05:
> Seems to be a low-voltage problem.

Yeah. The voltage from the expansion port is 4,15V. You need to use a voltage regulator (a "real" LDO) that handles that with sufficient margin.

mikkel75, 2023-12-05:
Actually I use a DC-DC from 4.15V to 3.3V - the issue was that I set the output to 3.0V (to save power, but still within ESP spec). The 3.0V somehow was too low for reliable UART RX.

ArnieO, 2023-12-05:
> Actually I use a DC-DC from 4.15V to 3.3V - the issue was that I set the output to 3.0V (to save power, but still within ESP spec). The 3.0V somehow was too low for reliable UART RX.

OK - that makes sense! To run at 3.0V during WiFi activity you will need a HUGE capacitor to ensure voltage does not drop too much. And parameters will drift over - so you were basically lucky it worked for some time. 
Glad you sorted it out!