UtilitechAS/amsreader-firmware #686: L2 phase current missing from version 2.2.24 with Aidon meter

JonathanJorstad, 2023-12-05:
After updating to 2.2.24, the L2 phase current is always displayed as zero.

**Expected behavior**
L2 phase current is displayed or calculated in the case the Aidon meter does not supply this reading.
[image]

**Screenshots**
[image]

**Hardware information:**
 - Country: Norway
 - Meter: Aidon
 - Encryption enabled no
 - AMS reader: Chip: esp8266 (80MHz) Device: [Pow-U (GPIO12)](http://192.168.10.103/vendor)
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.2.24
 - MQTT: no
 - MQTT payload type: [e.g. JSON]
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [e.g. 2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: yes

**Additional context**
I have heard that Aidon meters do not supply the L2 current, but in previous versions this current was indeed displayed

ArnieO, 2023-12-05:
Correct observation!

The calculation of L2 in the cases where the meter does not report it was in many situations quite incorrect, as it assumed balanced load (approximately same current on all 3 phases). 

As we have not found any way to reliably calculate L2 from available data, we decided to remove it from the display.

spacecosmos, 2023-12-06:
Just a thought, maybe the value should be "N/A*" (not available) or something similar? (And perhaps point out somewhere in a legend that * means not available?)

gskjold, 2023-12-06:
> Just a thought, maybe the value should be "N/A*" (not available) or something similar? (And perhaps point out somewhere in a legend that * means not available?)

I agree. Just as a note, the current behavior was not intended. I expected it to show data with a translucent bar labeled with "Estimated", but apparently the change in the calculation also made it always zero... I think we'll just go ahead and just remove it entirely in the next patch. I honestly don't understand why they expect us to calculate this phase, there is not enough data to do so

tronde-ams, 2023-12-06:
This is only relevant in Norway, so I will write in Norwegian.

*

Moderne mennesker aksepterer ikke at et spørsmål ikke kan ha et svar, så da er det bedre med en feil verdi for den strømmen som beviselig er der, enn ingen verdi... :-)

Fra spøk til alvor. 
Jeg rotet mye med dette for fire år siden, i håp om å komme frem til noe fornuftig. Jeg var optimistisk, helt til jeg fikk prøvd ut ideene, og sett litt på hva måleren faktisk forteller meg.

1: Strøm og spenning er snapshots som vi ikke aner om er fra samme tid som effekten var da de ble dyttet ut på HAN. Jeg mistenker at de ikke henger sammen tidsmessig. Dessuten er aldri linjespenningene helt like, spesielt ikke når det er høy last, og det er gjerne da vi har mest nytte av å kjenne I2 for å kunne balansere lasten.

2: Måleren gir ut data for reaktiv effekt eksport og import. Dette er nok et større problem enn om alt er fra samme tid. Det har vært vanlig å anta at reaktiv effekt er neglisjerbar for private husholdninger, og i den store sammenhengen (sett fra netteier) stemmer det nok. 

MEN - jeg har ingen problemer med å se import på opp mot 1000 VA, selv om 3-400 er mer vanlig. Jeg har ikke noe spennende i huset, og med slike verdier er vi godt over 1 A med strøm som vi ikke aner hvor ligger på linjene. I tillegg har jeg det siste året fått mye eksport, uten at noe er endret hos meg. Det varierer mellom 0 og ca. 350 VA. Det tyder på at netteier har satt inn kapasitiv last i nettet for å kompensere for induktiv last. 

Jeg har aldri sett at måleren viser både import og eksport samtidig, og det er mye som tyder på at den viser differansen mellom dem. Igjen aner vi ikke hvordan denne effekten er fordelt. Dette dreper egentlig veldig mye for kreative måter å beregne den manglende I2.

*

Hva jeg gjorde for fire år siden.

Jeg fant ut at det burde gå an å regne seg frem til noe brukbart, men det forutsetter at verdiene henger sammen tidsmessing OG at det ikke er reaktiv effekt inne i bildet. 

Inntil jeg begynte å se reaktiv eksport, trodde jeg at det gikk an å bruke den verdien som et flagg for tvilsom beregnet verdi, men nå som det er både import og eksport, og det ser ut som om det kun er differansen som vises, er det mindre gøy...

*

Utgangspunktet mitt var at jeg lagde et regneark med alle(?) mulige kombinasjoner av _prosentvis_ effekt mellom 0 og 100% 
Med aktiv bruk av slumptallsgeneratoren i Excel ende jeg opp med noe over 176000 linjer. Siden jeg var dum da jeg satte det opp, endte det opp med nær 240MB,

Jeg fant at det kunne være en slags sammenheng mellom den teoretiske I2 og forholdet mellom I1 og den teoretiske strømmen og I3 og den teoretiske strømmen.

Jeg delte opp i flere deler med ulike korreksjonsfaktorer som jeg lot Excel tegne opp som grafer, som igjen ble regnet om til polynomer. Nå har jeg glemt alle detaljene, men jeg kom frem til fire ulike korreksjonsfaktorer som ville holde feilen under ca. 15% og som ville vært mer en godt nok for å ikke sprenge sikringene.. 

MEN- nå er vi tilbake til den der reaktive effekten som ikke skulle vært der. Den kverker det beregnede forholdstallet mellom I1 og I3 og den symmetriske strømmen siden  vi ikke aner hvor den hører hjemme.

*

Dette kan egentlig like godt legges inn i et hjemeautomasjonssystem som leser måleren, som å ha det i Gunnars kode.
Jeg legger det som jeg har hatt gående i en arduinokode her, så kan folk prøve hvis de vil. Vær klar over at det ikke vil være korrekt, men det ser ut til å være bedre en å bare beregne den symmetriske verdien. Å presentere svaret med desimaler er helt meingsløst.

*
Jeg prøvde å formatere det som kode, med gikk heller dårlig så legger det i ei txt-fil. Dette er utdrag fra spaghettikoden min, så litt finpuss må til. Det er logikken som er interessant.

[aidon I2.txt](https://github.com/UtilitechAS/amsreader-firmware/files/13591940/aidon.I2.txt)

spacecosmos, 2023-12-06:
@tronde-ams, er det ikke slik målerne til f.eks.  **efergy** også fungerer? Altså at de simulerer eller regner ut den ene fasen, basert på de to andre?

tronde-ams, 2023-12-07:
> @tronde-ams, er det ikke slik målerne til f.eks. **efergy** også fungerer? Altså at de simulerer eller regner ut den ene fasen, basert på de to andre?

Jeg vet ikke hvordan de funker på 3-fase IT-nett som er vårt vanlige nett.

Hvis man skal få en verdi for manglende strøm som er rimelig nøyaktig, må man ha direkte tilgang til strømmens kurveform (bruke strømtrafo) og spenningenes kurveform. Da kan man jobbe med vektorer som er helt nødvendig. Jeg mistenker at målerne våre bruker digitale signalprossesorer som sampler alt sammen. Da koster det ikke mye, og er brukbart nøyaktig, for med mange samples av to strømmer og spenningene kan man oppnå det samme som med rene vektorer. Jeg har sett en del løsninger på nettet som bruker trafoer og likeretter, men det blir det ingen vektorer av.

HAN gir bare ut absoluttverdier, og det er ikke mulig å regne ut noe uten vektorer (eller mange nok samples som kan gjenskape dem). De som prøver å regne med formelen for symmetrisk nett, kommer ut med negative absoluttverdier for vekselstrøm noen ganger, og det er klin umulig å få negative absoluttverdier for vekselstrøm i virkeligheten. Jeg mistenker at Gunnar la inn sperre for negative svar en gang i fortiden, for tidlige versjoner ga ut negative strømmer nå og da.

Det jeg har prøvd på funker som en brukbar pekepinn hvis det er lite reaktivt, men det er ikke mulig å si hvor galt det blir i virkeligheten når reaktive strømmer blander seg inn, for de er der mer eller mindre hele tiden, og alle boliger er ulike med hensyn til last. Vi kan ha både reaktiv import på en fase, og eksport på en annen samtidig, uten at det vi ser på HAN sier noe om det. Da har vi i et problem.

Den opprinnelige krav-spekken for målerne sa at alle strømmer og spenninger skulle ut på HAN, nettopp for å kunne bruke lastbalansering, men noen sov i timen, og Aidon fikk godkjent sin løsning av en eller annen grunn. De bør ha nok data inni måleren til å regne ut I2 selv om de ikke måler den (de behøver den ikke), men de "har ingen planer om å gjøre den tilgjengelig" som de har fortalt noen en gang.

Det som er litt kinkig med å beregne noe, uten å vite hvor feil svaret kan være, er at man bruker svaret til noe det ikke burde bli brukt til. De som er fortrolig med måleteknikk klarer å forholde seg til at målinger kan være nokså feil, og fortsatt bruke det til noe fornuftig, mens de som ikke er det har en tendens til å legge alt for stor vekt på usikre verdier, og gjerne bruke mange desimaler også.

De gamle kjøkkenwattmetrene for 3-fas måler to strømmer som blir likerettet og sendt inn i et amperemeter som har en skala som er beregnet etter fast spenning (ex 220V). Jeg har ett på veggen, og ser helt tydelig at det noen ganger viser mye feil når jeg sammenlikner med den effekten som kommer vi HAN. Da er det helt tydelig hvor dårlig det funker å måle noe uten vektorer i et nett med ulik last.

Det som er mest viktig, er å ha en _ide_ om når I2 nærmer seg sikringens grense. Sikringer er dessuten nokså trege, så det er litt å gå på.

Strøm med desimaler er lite nyttig i virkeligheten, spesielt når det som kommer ut bare er et snapshot hvert 10. sekund. Jeg har målt alle linjestrømmene samtidig med strømtenger, og ser at det stort sett ikke er helt i tåka, men det er på langt nær "korrekt", og veldig avhengig av hva som er i huset.

Hvis noen vil prøve seg, bør det gå an å få et hjemmeautomasjonssystem til å regne slik jeg har vist med det som kommer fra HAN.

tronde-ams, 2023-12-07:
Jeg vet ikke hvorfor de ikke gir ut alle strømmene. Det vi vet, er at Aidon og Kaifa/Nuri helt sikkert benytter to-wattmetermetoden for å beregne effekt i målerne for IT-nettet. Da behøver de ikke mer enn to strømmer. 

[image]

https://illustrationprize.com/no/7-two-wattmeter-method-of-power-measurement.html

Aidon gir ikke ut den ene strømmen, mens Kaifa mangler den ene spenningen. Jeg vet ikke hvordan Kamstrup måler, men de fikk unntak fra kravet om hurtig oppdatering av effekten på HAN fordi de ikke klarte å få det til uten store endringer i hardware. Det kan kanskje tyde på at alle målerne har i underkant med regnekraft, og at Kaifa og Aidon proriterte oppdatering av effekt mest. 

Rent funksjonsmessig er jo energimålingen den primære funksjonen for målerne. Det andre er mer eller mindre "kjekt å ha", selv om det en gang fantes noen krav fra myndighetene som ingen av de tre opprinnelige leverandørene klarte å oppfylle fullt ut. 

Den tradisjonelle typen av dette wattmeteret var et mekanisk viserinstrument med to strømspoler og to spoler for spenning. De ble til sammen en analog elektromekanisk regnemaskin hvor strømmene og spenningenes øyeblikksverdier påvirket hverandre direkte med magnetfelt, og til slutt ga det en mekanisk kraft mot viseren som ble vist som watt. Nokså genialt, og veldig enkelt.

*

For dem som er norsk IP-adresse, og har lyst til å gå seg bort i vektorer, er dette linken til ei eldre bok i elektroteknikk for elkraftlinja på teknisk fagskole. Det går an å grave mye dypere i trefase enn det teknisk fagskole gjør, men det behøves ikke for å forstå at det som kommer ut på HAN er alt for lite for å gjenskape I2.

https://www.nb.no/items/7c47bc4cb6ee773c494c042186fff1e9?page=151

Det ligger mange godsaker hos nasjonalbiblioteket. Fysikken har ikke endret seg etter år 2000 som er grensen for gratis bøker, så gamle bøker om dette er helt OK med mindre man roter seg bort i noe om elektromagnetisme som bruker CGS-systemet og ikke SI.

tronde-ams, 2023-12-08:
Det er veldig mye å finne hos Nasjonalbiblioteket hvis man har norsk IP.
bokhylla.no

Ulempen er at søkefunksjonen er av typen "kan bli bedre". Det ligger masse yrkesskolebøker i alle mulige fag der også, for dem som føler for å lære noe nytt, eller friske opp gammel kunnskap.

Søk etter forfatter "Pelt, Th.P. van" , så dukker de andre bøkene i samme serie opp. Det finnes helt sikkert noe fra andre forfattere også.

*

Når det gjelder trefase, er jo vi i Norge spesielle siden vi har nesten kun 230V IT (trekantkoplet), mens resten av verden stort sett har 400V TV (stjernekoplet) nett. Nord-Amerika skiller seg ut med flere (for oss) merkelige systemer.

Nå brukes faktisk IT i de aller fleste land, men ikke for forbrukere. IT har den egenskapen at det ikke går ned ved en (1) kortslutning mellom fase og jord slik TN gjør. Denne egenskapen gjør at IT er vanlig i sykehus og deler av kjemisk industri hvor fungerende nett noen ganger er viktigere enn sikkerhet. Det krever mer overvåking. Disse nettene er satt opp lokalt i bygget med egen trafo. Siden IT ikke er vanlig, er det heller ikke så mye å lese om det i lærebøker. 

Når man skal prøve å finne ut av trefase, bør man være veldig forsiktig med å hente informasjon fra nettet, for mye der er basert på det de bruker i Nord-Amerika, og fullstendig feil for oss i Norge.

De aller fleste norske bøkene tar utgangspunkt i det en vanlig elektriker møter, og det er symmetrisk last, typisk en trefasemotor som skal ha en passende ledningsføring og sikringer for denne. Usymmetrisk last er mye mer aktuelt å bry seg om for nettselskapene, enn for elektrikere, og det er få som jobber der, så tilgangen på lesestoff er mye mindre.

closed 2023-12-23 by gskjold