UtilitechAS/amsreader-firmware #688: Low prio bug: Tariff thresholds visible even though not a Norwegian region on latest 2.2.24

zelbanna, 2023-12-06:
**Describe the bug**
According to description/help the Tariff thresholds should only be visible for Norwegian regions - I am on SE3 and can see it

**Expected behavior**
Shouldn't be visible on my screen

**Relevant firmware information:**
 - Version: 2.2.24
 - ENTSO-E API enabled: no
 -  Fetch prices enabled: yes
 -  Region SE3

gskjold, 2023-12-15:
See #680, feel free to comment

zelbanna, 2023-12-16:
Sure, just like in Norway there is obviously not a single "in Sweden" but loads of different power companies with different models, just in stockholm where I live there are companies with both peak- and non-peak based tariffs (for the entire year but different for different months)

This peak hour usage model is hard to generalize... maybe if one can define a "records peak hour consumption between x-y hours" as an optional field. E.g. in our case - where I live - this peak is  recorded between 07-19 on weekdays. The value is then calculated as the mean value of the top 3 readings during the month, times 100 SEK during winter months and 50 SEK during summer months (yes, they really like to make it complicated)

ArnieO, 2023-12-18:
> Sure, just like in Norway there is obviously not a single "in Sweden" but loads of different power companies with different models, just in stockholm where I live there are companies with both peak- [...]

Yes, it is a big mess. A BIG mess! This is why we are very reluctant to try to implement something more on our side.

The idea that power tarrifs implemented in this way should somehow work (actually contribute to adjusting the load on the grid) is just ludicrous.