UtilitechAS/amsreader-firmware #690: Total meeter consumption

eireit, 2023-12-07:
**Describe the bug**
After updating to 2.2.24 the Accumulated import is swapping between total meeter consumption and 0. And then count upwards...
After some time it changes back to the total consumption. When reboot it starts from 0 again...

**To Reproduce**
Steps to reproduce the behavior:
Logg the value of "ic"

**Expected behavior**
A clear description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware information:**
 - Country: Norway
 - Meter: Kaifa MA304H3E
 - Encryption enabled: no
 - AMS reader: [e.g. Pow-U, esp32s2 (160MHz)
 
**Relevant firmware information:**
 - Version: 2.2.24
 - MQTT: yes
 - MQTT payload type: JSON
 - HAN baud and parity: 2400, 8E1
- ENTSO-E API enabled: yes

**Additional context**

ArnieO, 2023-12-08:
Strange - and unexpected!
Can you try to roll back to v2.2.23 and check that this is indeed related to update from v2.2.23 to v2.2.24?

gskjold, 2023-12-08:
The accumulated values will always start at 0 on a reboot, and since Norwegian meters only report accumulated values once per hour, it will keep doing so until it gets the correct data. If you see this happen during operation, my guess is that the device reboots.

eireit, 2023-12-08:
@gskjold gskjold, I think thats the problem. It has started to do reboots, several times a day. 
This has happend after firmware update (2.2.24), any tips to fix this issue?

ArnieO, 2023-12-09:
> @gskjold gskjold, I think thats the problem. It has started to do reboots, several times a day. This has happend after firmware update (2.2.24), any tips to fix this issue?

Which hardware are you running?

eireit, 2023-12-09:
Meter: Kaifa MA304H3E
Encryption enabled: no
AMS reader: [e.g. Pow-U, esp32s2 (160MHz)

Downgraded to 2.2.23 and it has not rebooted for 12hours.

ArnieO, 2023-12-10:
> Downgraded to 2.2.23 and it has not rebooted for 12hours.

Interesting, thank you for reporting!

gskjold, 2023-12-21:
I am honestly not sure what is going on here, could you export a configuration file and send me? gunnar@amsleser.no

gskjold, 2023-12-23:
Cant really see anything wrong with your config. Please try v2.2.25 and untick "Allow 802.11b legacy rates"

eireit, 2023-12-26:
Hi
Upgraded to v2.2.28 and unchecked the allow 802.11b legacy. The it started to reboot again after approx 1 houer (Software reset). I have downgraded to v2.2.27 and will do a similar check.

BR eirik

gskjold, 2024-01-26:
Any updates on this issue? How is 2.2.28 treating you?

eireit, 2024-01-29:
Hi, has been stable for over 2 days. looks good...

Br Eirik

closed 2024-03-21 by gskjold