UtilitechAS/amsreader-firmware #694: How to connect the temp sensor DS18B20 to Pow-K+

ielthbe, 2023-12-14:
How to connect the temp sensor DS18B20 to Pow-K+ 
[image]
[image]

ArnieO, 2023-12-14:
Well done!
Here is a sneak peek on the next version Pow-K that we will start selling when current stock is sold out, expected somewhere in Q1/24. We decided to include a footprint for DS18B20, and included the pullup resistor (R12; costs almost nothing), so that interested users can easily add a DS18B20. It is a relatively expensive component (unless you buy a cheap Chinese clone), so we think it is better that those who need it install it themselves.
[image]

gskjold, 2023-12-15:
I might be wrong here, but using "+" from the external power source, doesn't that also apply +4.5 (or +5v in case of USB power) to the GPIO pin? Could burn out the GPIO?

ArnieO, 2023-12-18:
> I might be wrong here, but using "+" from the external power source, doesn't that also apply +4.5 (or +5v in case of USB power) to the GPIO pin? Could burn out the GPIO?

I assume he is using a pullup resistor between 5V and GPIO.

This is how the optional sensor is implemented on the coming version of Pow-K:
[image]

closed 2023-12-21 by gskjold