UtilitechAS/amsreader-firmware #699: Multipliers ignored in P1 mode, and missing data.

otsafe, 2023-12-21:
**Describe the bug**
We are testing out P1 in Norway and have the following issues:

Multipliers ignored in P1 mode.

Also some data is unknown
Manufacturer: Unknown
Model: unknown
ID: unknown

**To Reproduce**
Steps to reproduce the behavior:
Change multiplier and look at front page, values remains the same. Also tried to reboot.
The board is modified by bridging the pins to receive P1 data, according to documentation.

**Expected behavior**
Same as in HAN mode.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware information:**
 - Country: [Norway]
 - Meter: [Kamstrup]
 - Encryption enabled [no]
 - AMS reader: [Pow-K+]
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: [e.g. 2.2.24]
 - MQTT: [no]
 - MQTT payload type: [-]
 - HAN GPIO: []
 - HAN baud and parity: [115200 8n1 (Buffer: 512)]
 - Temperature sensors []
 - ENTSO-E API enabled: [yes]

**Additional context**
Looks like it receives correct data. Voltage levels appears ok. Displayed Import W is 60x lower than expected. This meter is 3phase IT and using a current transformer and is operating in x60 mode. Before switching to P1 mode, it worked fine with the following multipliers:
W: 60, V:1, Amp: 6, kWh: 60

```
(D) Received valid DSMR at 0
(V) (AmsDataStorage) Day data timestamp age 1703158944 - 1703145607 > 3600
(D) Its time to update data storage
(V) (EnergyAccounting) Adding 0.0000 kWh import
(V) (EnergyAccounting)  and 0.0000 NOK
(V) (EnergyAccounting)  calculating threshold, currently at 0
(V) (EnergyAccounting)  new threshold 0
(D) Serving /data.json over http...
(D) Serving /data.json over http...
(V) DSMR frame:
(V) 2F 4B 41 4D  35 0D 0A 0D  0A 30 2D 30  3A 31 2E 30
(V) 2E 30 28 32  33 31 32 32  31 31 32 34  32 33 33 57
(V) 29 0D 0A 31  2D 30 3A 31  2E 38 2E 30  28 30 30 30
(V) 30 39 32 32  32 2E 34 37  31 2A 6B 57  68 29 0D 0A
(V) 31 2D 30 3A  32 2E 38 2E  30 28 30 30  30 30 30 30
(V) 30 30 2E 30  31 37 2A 6B  57 68 29 0D  0A 31 2D 30
(V) 3A 33 2E 38  2E 30 28 30  30 30 30 30  33 36 39 2E
(V) 34 32 39 2A  6B 56 41 72  68 29 0D 0A  31 2D 30 3A
(V) 34 2E 38 2E  30 28 30 30  30 30 35 36  38 36 2E 39
(V) 31 36 2A 6B  56 41 72 68  29 0D 0A 31  2D 30 3A 31
(V) 2E 37 2E 30  28 30 30 30  30 2E 30 31  34 2A 6B 57
(V) 29 0D 0A 31  2D 30 3A 32  2E 37 2E 30  28 30 30 30
(V) 30 2E 30 30  30 2A 6B 57  29 0D 0A 31  2D 30 3A 33
(V) 2E 37 2E 30  28 30 30 30  30 2E 30 32  32 2A 6B 56
(V) 41 72 29 0D  0A 31 2D 30  3A 34 2E 37  2E 30 28 30
(V) 30 30 30 2E  30 30 30 2A  6B 56 41 72  29 0D 0A 31
(V) 2D 30 3A 32  31 2E 37 2E  30 28 30 30  30 30 2E 30
(V) 32 31 2A 6B  57 29 0D 0A  31 2D 30 3A  34 31 2E 37
(V) 2E 30 28 30  30 30 30 2E  30 30 30 2A  6B 57 29 0D
(V) 0A 31 2D 30  3A 36 31 2E  37 2E 30 28  30 30 30 30
(V) 2E 30 30 30  2A 6B 57 29  0D 0A 31 2D  30 3A 32 32
(V) 2E 37 2E 30  28 30 30 30  30 2E 30 30  30 2A 6B 57
(V) 29 0D 0A 31  2D 30 3A 34  32 2E 37 2E  30 28 30 30
(V) 30 30 2E 30  30 30 2A 6B  57 29 0D 0A  31 2D 30 3A
(V) 36 32 2E 37  2E 30 28 30  30 30 30 2E  30 30 37 2A
(V) 6B 57 29 0D  0A 31 2D 30  3A 32 33 2E  37 2E 30 28
(V) 30 30 30 30  2E 30 31 33  2A 6B 56 41  72 29 0D 0A
(V) 31 2D 30 3A  34 33 2E 37  2E 30 28 30  30 30 30 2E
(V) 30 30 30 2A  6B 56 41 72  29 0D 0A 31  2D 30 3A 36
(V) 33 2E 37 2E  30 28 30 30  30 30 2E 30  30 39 2A 6B
(V) 56 41 72 29  0D 0A 31 2D  30 3A 32 34  2E 37 2E 30
(V) 28 30 30 30  30 2E 30 30  30 2A 6B 56  41 72 29 0D
(V) 0A 31 2D 30  3A 34 34 2E  37 2E 30 28  30 30 30 30
(V) 2E 30 30 30  2A 6B 56 41  72 29 0D 0A  31 2D 30 3A
(V) 36 34 2E 37  2E 30 28 30  30 30 30 2E  30 30 30 2A
(V) 6B 56 41 72  29 0D 0A 31  2D 30 3A 33  32 2E 37 2E
(V) 30 28 32 34  34 2E 32 2A  56 29 0D 0A  31 2D 30 3A
(V) 35 32 2E 37  2E 30 28 32  34 33 2E 36  2A 56 29 0D
(V) 0A 31 2D 30  3A 37 32 2E  37 2E 30 28  32 34 34 2E
(V) 32 2A 56 29  0D 0A 31 2D  30 3A 33 31  2E 37 2E 30
(V) 28 30 30 30  2E 31 2A 41  29 0D 0A 31  2D 30 3A 35
(V) 31 2E 37 2E  30 28 30 30  30 2E 30 2A  41 29 0D 0A
(V) 31 2D 30 3A  37 31 2E 37  2E 30 28 30  30 30 2E 30
(V) 2A 41 29 0D  0A 21 37 44  44 31 0D 0A
```
The consumption on this meter is almost static, and should be around 840W all the time.

[image]

ArnieO, 2023-12-21:
Interesting! This is as far as I know the first known test of P1 smart meters in Norway. I think we were already in dialogue by email. 

- Which Kamstrup model is this? A photo would be appreciated. 
- Please provide screendump of the info-page (where you indicate manufacturer/model is missing). 

We will analyze the provided debug and revert.

ArnieO, 2023-12-21:
OK, this is ASCII - so it is easy to decode.
Below is the content of the payload you provided, the first part of each line is the OBIS code identifying each parameter. 

So the firmware does not identify the meter correctly. Is this a "production model" - or is it a "test model" that may have a non-standard text in the first two lines?

Total accumulated power is displayed incorrectly (9222.5 kWh). Did you do something with the multipliers? It would be useful to see your config settings.

14W import power seems to be correctly displayed.

```
/KAM5
0-0:1.0.0(231221124233W)
1-0:1.8.0(0000922.471*kWh)
1-0:2.8.0(0000000.017*kWh)
1-0:3.8.0(00000369.429*kVArh)
1-0:4.8.0(000005686.916*kVArh)
1-0:1.7.0(0000.014*kW)
1-0:2.7.0(0000.000*kW)
1-0:3.7.0(0000.022*kVAr)
1-0:4.7.0(0000.000*kVAr)
1-0:21.7.0(0000.021*kW)
1-0:41.7.0(0000.000*kW)
1-0:61.7.0(0000.000*kW)
1-0:22.7.0(0000.000*kW)
1-0:42.7.0(0000.000*kW)
1-0:62.7.0(0000.007*kW)
1-0:23.7.0(0000.013*kVAr)
1-0:43.7.0(0000.000*kVAr)
1-0:63.7.0(0000.009*kVAr)
1-0:24.7.0(0000.000*kVAr)
1-0:44.7.0(0000.000*kVAr)
1-0:64.7.0(0000.000*kVAr)
1-0:32.7.0(244.2*V)
1-0:52.7.0(243.6*V)
1-0:72.7.0(244.2*V)
1-0:31.7.0(000.1*A)
1-0:51.7.0(000.0*A)
1-0:71.7.0(000.0*A)
```

otsafe, 2023-12-21:
Yes, that is true. Got it activated today.

The meter is probably sending 14W, but since this this meter is configured with a multiplier it should actually be 60 x 14W = 840W. The screen shows the raw value from the meter, but ignores the multiplier that is set to x60 for W. This worked fine in HAN mode before the protocol was switched.

The meter is a production meter. But since we probably are the first, it could be that some of the pushed data is not correct. If you suggest it should be something else I can forward it to them.

[image]

Serial numbers removed:
[image]

[image]

gskjold, 2023-12-21:
@ArnieO not sure what you did, but when I convert the hex, I get 9222kwh, not 922:

```
/KAM5

0-0:1.0.0(231221124233W)
1-0:1.8.0(00009222.471*kWh)
1-0:2.8.0(00000000.017*kWh)
1-0:3.8.0(00000369.429*kVArh)
1-0:4.8.0(00005686.916*kVArh)
1-0:1.7.0(0000.014*kW)
1-0:2.7.0(0000.000*kW)
1-0:3.7.0(0000.022*kVAr)
1-0:4.7.0(0000.000*kVAr)
1-0:21.7.0(0000.021*kW)
1-0:41.7.0(0000.000*kW)
1-0:61.7.0(0000.000*kW)
1-0:22.7.0(0000.000*kW)
1-0:42.7.0(0000.000*kW)
1-0:62.7.0(0000.007*kW)
1-0:23.7.0(0000.013*kVAr)
1-0:43.7.0(0000.000*kVAr)
1-0:63.7.0(0000.009*kVAr)
1-0:24.7.0(0000.000*kVAr)
1-0:44.7.0(0000.000*kVAr)
1-0:64.7.0(0000.000*kVAr)
1-0:32.7.0(244.2*V)
1-0:52.7.0(243.6*V)
1-0:72.7.0(244.2*V)
1-0:31.7.0(000.1*A)
1-0:51.7.0(000.0*A)
1-0:71.7.0(000.0*A)
!7DD1
```

This will be correct when I add multipliers for DSMR payload

ArnieO, 2023-12-21:
> not sure what you did, but when I convert the hex, I get 9222kwh, not 922

I trusted Bing.com (AI) to convert it!  😲 
That was a surprising AI-error, but good to see the error was not with us. 

> Model and ID is not reported in the payload, so that cannot be known. Will fix manufacturer

Is model and serial number (ID) normally part of the OBIS list, or is it in outside the DSMR frame?
In the [Kamstrup documentation for the HAN-P1 module](https://documentation.kamstrup.com/docs/HAN_P1/en-GB/Data_sheet/CONTE72CCF442CC34A4FAC97443A71094C63/), model and ID are not part of the payload.

gskjold, 2023-12-21:
Model and ID is normally part of the payload, but I very rarely see it in DSMR payloads, so it is not unexpected.

otsafe, 2023-12-21:
Here is a mockup of an old screenshot when running in P1 vs HAN mode. For energy, multiplying with 60 gives values correspond with earlier readings. Hope it helps. Load is about the same as now, since it is a standby supply with only a transformer connected to it.

[image]

gskjold, 2023-12-21:
Test firmware with fix:
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13743185/esp32s2.zip)

otsafe, 2023-12-21:
Thanks. Great.
Should I flash the firmware.bin OTA, and ignore the other files? I only have remote access.

ArnieO, 2023-12-21:
Yes, only firmware.bin.
Follow the instructions here to upload to your device: https://github.com/UtilitechAS/amsreader-firmware/wiki/Flashing-via-web-GUI

otsafe, 2023-12-21:
Thanks, now looks correct. What a great response here...

[image]

I see that the current is decoded to this:
1-0:31.7.0(000.1*A)
1-0:51.7.0(000.0*A)
1-0:71.7.0(000.0*A)

I'm uncertain whether the current should be multiplied by a factor of x6 or x60. Prior to transitioning to P1, it was set at x6, and I observed readings such as 5.xxA and 3.xxA (refer to the previous image above). The data decoded by @ArnieO indicates a minimum current of 000.1A. This implies that the lowest measurable current is 0.1A, which is equivalent to either 0.6A or 6.0A.

Could you please confirm whether the hexadecimal code contains additional decimal places for the current value? Given that the real power is 840W, a current of 0.6A would not suffice. Therefore, it's probable that the correct current multiplier should indeed be x60, unless there's a possibility of losing some decimal precision during the parsing process.

The load is reactive (100kVA transformer running without load), so it gives some weird numbers. I can try to call Norgesnett tomorrow to verify what multiplier should used on the current in P1 format.

Can you please share the script or code you used to parse the hex values?

gskjold, 2023-12-22:
The hex you get is a plain ascii string which decodes to the characters you describe, there is no script involved. I suspect 6 is the correct multiplier here, but because the payload only has one decimal for amps, accuracy becomes an issue.

ArnieO, 2023-12-22:
@otsafe You probably already considered this, but I think it would be easier for you if you could increase the load during your testing?

otsafe, 2023-12-22:
@gskjold Thanks, you are right. I ran it through an online hex to ascii converter and got the output posted here. I'm not familiar with this P1 format, so I thought it required some decoding...

@ArnieO I'll see what I can do, but it is complicated. It is backup power for another supply with automatic failover. I have asked it they can remove multiplicator for current since P1 has 3 digits, it should not be required. We will never load it with 999.9A.

Thank you very much for now.

closed 2023-12-23 by gskjold