UtilitechAS/amsreader-firmware #701: Release v2.2.25 causing boot loop on esp8266

Belaial, 2023-12-23:
**Describe the bug**
I upgraded my AMS reader to v2.2.25 this morning via the GUI, shortly after I noticed the up-time kept resetting in GUI, via ping I could see it lost connection around 5-10 seconds and then responded for about 5-10 seconds.

I was able to downgrade via GUI to 2.2.24 and it's now 100% stable again.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to web GUI and upgrade the device to 2.2.25
2. Wait for reboot and see if it's stable or not

**Expected behavior**
-

**Screenshots**
[image]

**Hardware information:**
 - Country: Finland (but using prices from Sweden)
 - Meter: Kamstrup
 - Encryption enabled: No
 - AMS reader: Chip: esp8266 (80MHz) / Device: [Wemos D1 mini]
 - M-bus adapter (if applicable): https://www.ebay.com/itm/274713581754

**Relevant firmware information:**
 - Version: 2.2.25
 - MQTT: Yes
 - MQTT payload type: HomeAssistant
 - HAN GPIO: GPIO5
 - HAN baud and parity: 2400 8E1
 - Temperature sensors: No
 - ENTSO-E API enabled: No (I don't know what this is even)

**Additional context**
-

Belaial, 2023-12-23:
Forgot to add that rebooting the AMS reader after upgrade did not help for me, it just started to boot loop again.

Hawkeye4060, 2023-12-23:
@Belaial: How did you manage to downgrade?

Belaial, 2023-12-23:
@Hawkeye4060 

I downloaded the 2.2.24 firmware in .bin format from - https://github.com/UtilitechAS/amsreader-firmware/releases/tag/v2.2.24
(In my case ams2mqtt-esp8266-2.2.24.bin depends on the hardware you have what file you should use)

I then logged in to the web GUI on the AMS reader and found under the "information icon" (middle one in this image) that you are able to upload a firmware file manually
[image]

I those the option "Select firmware file for upgrade" and uploaded the ams2mqtt-esp8266-2.2.24.bin file as soon as it responded to ping again.
[image]

It stayed online while the upload / downgrade to 2.2.24 was performed, I then waited for the downgrade to complete, it did not take long, 30-60 seconds maybe and then I checked it had downgraded to 2.2.24 and monitored it via ping to see that is was stable again.

tbergo, 2023-12-23:
Same issue here.

Observation:
Looks like there are some MQTT issues. MQTT "sign" in the GUI header stays yellow on release v.2.2.25. After downgrade to release v2.2.24, everything are OK and MQTT "sign" turns green.
On release v2.2.25 the debug log shows that the AMS reader connects to the MQTT server, then the AMS reader reboots. 
This behaviour looping endlessly.

Belaial, 2023-12-23:
@tbergo 

I did not remember to check debug so good that you did that, I noticed the same thing with MQTT I think, I mean the yellow MQTT "sign", not sure tho as I downgraded again quite fast, to close to Christmas so not much time to fiddle around with home automation :smile:

tbergo, 2023-12-23:
From the MQTT server log:
Mosquitto broker
2023-12-23 10:04:06: New connection from 192.168.1.211:57329 on port 1883.
2023-12-23 10:04:06: New client connected from 192.168.1.211:57329 as AMSreader (p2, c1, k10, u'x').
2023-12-23 10:04:20: Client AMSreader closed its connection.
2023-12-23 10:04:27: New connection from 192.168.1.211:65435 on port 1883.
2023-12-23 10:04:27: New client connected from 192.168.1.211:65435 as AMSreader (p2, c1, k10, u'x').
2023-12-23 10:04:40: Client AMSreader closed its connection.
2023-12-23 10:04:48: New connection from 192.168.1.211:56874 on port 1883.
2023-12-23 10:04:48: New client connected from 192.168.1.211:56874 as AMSreader (p2, c1, k10, u'x').
2023-12-23 10:05:01: Client AMSreader closed its connection.

gskjold, 2023-12-23:
I have pulled 8266 from the release. Let me know if someone gets this behavior on esp32. Will look into fix

Hawkeye4060, 2023-12-23:
@Belaial
Thank you for your help and explanation.
It now works again with v2.2.24.

@gskjold 
My version is the ESP8266.

ArnieO, 2023-12-23:
We received a message in the webshop today on the same issue. That was also a product with ESP8266.

gskjold, 2023-12-23:
Did you manage to guide them to revert? As a tip I would unplug from meter and disable mqtt to make it stop before downgrade

ArnieO, 2023-12-23:
> Did you manage to guide them to revert? As a tip I would unplug from meter and disable mqtt to make it stop before downgrade

I referred to this thread and included link to https://github.com/UtilitechAS/amsreader-firmware/issues/701#issuecomment-1868251479

Belaial, 2023-12-23:
> @Belaial Thank you for your help and explanation. It now works again with v2.2.24.

Your welcome, happy I could help!

Belaial, 2023-12-23:
> I have pulled 8266 from the release. Let me know if someone gets this behavior on esp32. Will look into fix

Sounds good, feel free to close this ticket unless you want to keep it around for more troubleshooting / testing reports.

gskjold, 2023-12-23:
Please test v2.2.26

Hawkeye4060, 2023-12-23:
V2.2.26 didn't help.
Still no HAN communication.

Hawkeye4060, 2023-12-23:
Some other strange things happens.
I downgraded to v2.2.24, and this works for about 4 hours, and then the HAN-communication stopped, and now it doesn't matter what firmware I upload to the module.
I have tried v2.2.23 but still no HAN.
Any suggestions?

Hawkeye4060, 2023-12-23:
V2.2.27 doesn't fix my issue either ☹

Belaial, 2023-12-25:
Will be AFK a few more days due to Christmas and so on, no time to test for now, will get back once I have time unless someone else can test before me, I see @Hawkeye4060 have tested already but good if more have time to test.

Hawkeye4060, 2023-12-25:
@Belaial: I think I have another issue. See this thread: https://github.com/UtilitechAS/amsreader-firmware/issues/706#issue-2055049182

Belaial, 2023-12-27:
@gskjold - Today I took the time to upgrade from 2.2.24 ---> 2.2.28, 2+ hours up time now and all is working fine for me so far, great work!

@Hawkeye4060 - I checked https://github.com/UtilitechAS/amsreader-firmware/issues/706#issue-2055049182 but I can't say I see the same issue, my HAN-communication seems stable for 2+ hours now, I don't however use the same reader since I see in the ticket you use the "AMS reader: Pow-K", I built my own so maybe this is only an issue for your model.

Belaial, 2023-12-29:
2+ days up time now, I see no issue with 2.2.28 and my reader based on esp8266.
@gskjold feel free to close this ticket unless there is some more troubleshooting needed.