UtilitechAS/amsreader-firmware #702: Version v2.2.25 does not work on my Aidon 6534

niklaslj, 2023-12-23:
**Describe the bug**
Uppgrading from **v2.2.23** to **v2.2.23** this morning. Now it can not read my meter Aidon 6534.
Getting HAN-error. Singel red flash and various error messages indicating read error ie "Serial Parity Error", "Missing frame char..".

Reloaded **v2.2.23** and the amsreader works again.

**To Reproduce**
Steps to reproduce the behavior:
Upgrade to v2.2.25 and connect to an Aidon 6535 meter.

**Hardware information:**
 - Country: Sweden
 - Meter: Aidon 6534 (6484 RJ45 interface)
 - Encryption enabled no
 - AMS reader: Pow-U+, esp32s2 (160MHz)
 - M-bus adapter: no

**Relevant firmware information:**
 - Version: 2.2.25
 - MQTT: no
 - MQTT payload type: 
 - HAN GPIO: 
 - HAN baud and parity: 2400 8E1
 - ENTSO-E API enabled: no?

niklaslj, 2023-12-23:
Uploading **v2.2.24** works fine!
Tried v2.2.25 again but same error.

gskjold, 2023-12-23:
What is the buffer size set to?

niklaslj, 2023-12-23:
256 (default value, not changed)

gskjold, 2023-12-23:
Please test v2.2.26 
If you have any issues, try increasing buffer size. I'm guessing 768 should be a good fit for you

niklaslj, 2023-12-24:
Changed  buffer size to 768 and upgraded to v2.2.28.
Works fine!!

God Jul!
/Niklas

closed 2023-12-24 by niklaslj