UtilitechAS/amsreader-firmware #703: Amperage L2 missing on 2.2.26 Chip: esp32s2 , Device: Pow-K+

nesegrus, 2023-12-23:
Just upgraded from 2.2.25 to 2.2.26 and after upgrade the phase current L2 is missing. Tried reboot several times - did not fix the issue.
Then downgraded to 2.2.25 and everything works ok again. 

My device is Pow-K+ and the chip is  esp32s2.
The model of the meeter is  :  
Manufacturer: Kamstrup
Model: 6841121BN245101040

gskjold, 2023-12-23:
Dang it, sorry about that, some test code ended up in the release.... New one coming up

A-Wangen, 2023-12-24:
hi, i also experience the same issue on l2 on FW V2.2.28 with esp32s2.

gskjold, 2023-12-24:
What meter manufacturer and model?

A-Wangen, 2023-12-24:
> What meter manufacturer and model?

updated info above

gskjold, 2023-12-24:
This is a IT meter that does not report I2. We have previously reported a calculated value for I2 for these meters, but because the calculation can be wildly off target, we have decided to remove this calculation

closed 2024-01-26 by gskjold