UtilitechAS/amsreader-firmware #704: Homeassistant missing some Sensors (MQTT values) from ESP32 with 2.2.27

gvenaas, 2023-12-23:
Some MQTT messages are lost in Homeassistant with 2.2.27. Reverted to 2.2.25 and all is OK

EX: AMS reader Active import

Chip: esp32 (160MHz)
Device: [Wemos LOLIN D32]

Manufacturer: Kamstrup
Model: 6841121BN243101040

ENTSOE : On

daroc, 2023-12-24:
Also had MQTT issues with the new firmware, everything works when reverted back to 2.2.25. 

Device information
Chip: esp32s2 (160MHz)
Device: [Pow-K+]

Meter
Manufacturer: Kamstrup
Model: 6841121BN243101040

cmock, 2023-12-24:
Same here. 2.2.26 and 2.2.27 seem to be affected, from what I see those are only sending the .../prices and .../state topics, .../power, .../realtime, .../energy are missing.

ESP32, MQTT type is HomeAssistant.

gskjold, 2023-12-24:
Fixed in 2.2.28!

cmock, 2023-12-24:
That fixed it for me, thanks! (And now go and enjoy the holidays :-)

gvenaas, 2023-12-25:
Works for mee as well....

closed 2023-12-25 by gvenaas