UtilitechAS/amsreader-firmware #706: No HAN communication after update

Hawkeye4060, 2023-12-24:
I have been using the Pow-K for more than a year, and updated every time, there were a new firmware.
But after updating from v2.2.24 to v2.2.25 the HAN communication has failed.
I downgraded to v2.2.24 again which then worked again for about 4 hours, but then the communication stopped.
I have tried to update to 26, 27 and 28, but none of them worked.
Now I have tried to downgrade to 23 and 24 but the communication is totally dead.
with v2.2.23 I have also done a Factory Reset, but the only thing that works is the prices and webserver.

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup Omnipower
 - Encryption enabled yes
 - AMS reader: Pow-K

**Relevant firmware information:**
 - Version: 2.2.23 to 2.2.28
 - MQTT: yes
 - MQTT payload type: JSON
 - HAN GPIO: ??default??? can't find any information about this???
 - HAN baud and parity: 2400 8E1
 - Temperature sensors: none
 - ENTSO-E API enabled: ??

best regards Mogens

gskjold, 2023-12-24:
This happens with Kamstrup from time to time. Meter needs to be restarted. Last incident here: https://github.com/UtilitechAS/amsreader-firmware/issues/644

Hawkeye4060, 2023-12-24:
Thank you, Gunnar.
I will contact my supplier to get this fixed.

Glædelig jul til dig og dine, og tak for god support :-)
mvh Mogens

gskjold, 2023-12-24:
No problem! God jul :)

niklaslj, 2023-12-24:
Had the same problem after upgrading to 2.2.25. Solved by changing the buffer size from 256 to 768.
Hope this works for you as well.
God Jul!

ArnieO, 2023-12-24:
@Hawkeye4060 You only need to contact grid company if meter stopped sending data.
If you see blue blinking on the LED every 10 seconds, the meter is still sending data.

Hawkeye4060, 2023-12-24:
The buffer can only be 256. It is the ESP8266 version I have.
There is no blue flash, so I think it has stopped sending data.

God Jul til alle.

Hawkeye4060, 2023-12-27:
Kamstrup has reset my meter, and the HAN-com is working again.

closed 2023-12-27 by Hawkeye4060

ArnieO, 2023-12-27:
> Kamstrup has reset my meter, and the HAN-com is working again.

Good to hear! 👍

ArnieO, 2024-02-29:
We issued a blog post today with information on how the Kamstrup Omnipower issue ("No HAN data") can be remedied. 
https://www.amsleser.no/module/ets_blog/blog?id_post=33