UtilitechAS/amsreader-firmware #707: Unknown data from kamstrup meter.

effix1983, 2023-12-25:
From the HAN interface I get "Unknown data received". after a while it says  "Serial FIFO overflow". If i HAN baud / parity to any other setting it just fails and recieves no data.

**Hardware information:**
 - Country: Denmark
 - Meter: Kampstrup (684-11-31B-N14-3101-010)
 - Encryption enabled: no
 - AMS reader: ESP32
 - M-bus adapter (if applicable):

 - Version: v2.2.28
 - MQTT: yes
 - MQTT payload type: JSON
 - HAN GPIO: UART0
 - HAN baud and parity: Autodetect
 - ENTSO-E API enabled: no

Readout from telnet
*** Remote debug - over telnet - for ESP32 - version 3.0.5
* Host name: ams-f071 IP:192.168.50.168 Mac address:94:E6:86:3B:71:F0
* Free Heap RAM: 190164
* ESP SDK version: 4.4.5.230722
******************************************************

* Please type the command and press enter to execute.(? or h for this help)
***
(W) No HAN data received last 30s, single blink
(E) Ended up in default case while unwrapping...(tag BD)
(W) Unknown data received
(V)   payload:
(V) BD 48 05 6A  FF 6E 20 64  65 BA A1 A5  B1 C1 C1 A5
(V) 42 A2 85 9D  42 44 29 0D  0D 0A 1B 5B  30 16 56 A8
(V) 48
(E) Ended up in default case while unwrapping...(tag 70)
(W) Unknown data received
(V)   payload:
(V) 70 61 79 6C  0A 1B 5B 30  6D 28 56 29  20 42 44 20
(V) 34 38 41 A2  A2 02 92 20  30 44 20 30  41 20 31 42
(V) 20 35 42 20  20 33 30 20  31 36 20 35  36 20 41 38
(V) 20 0D 0D 0A  1B 5B 30 6D  A3 6A FF 94  95 91 6E 20
(V) 64 65 1A 85  CD 95 77 68  69 6C 65 20  75 6E 77 72
(V) 61 70 70 BA  82 4A 02 B9  AD B9 BD DD  B9 64
(E) Ended up in default case while unwrapping...(tag 30)
(W) Unknown data received
(V)   payload:
(V) 30 20 36 31  02 0D 02 02  A2 02 92 82  02 9A 82 02
(V) 92 82 02 9A  6D 28 56 82  20 31 42 20  35 39 31 20
(V) 36 A8 48 68  B4 20 36 43  20 02 BA AA  02 A2 20 42
(V) 20 36 1B 5B  30 6D 64 65  66 61 57 29  20 55 64 61
(V) 74 43 A1 A3
(E) Ended up in default case while unwrapping...(tag 0B)
(W) Unknown data received
(V)   payload:
(V) 0B 02 20 33  36 20 33 31  20 20 30 32  20 30 44 20
(V) 30 32 20 30  32 38 32 20  A1 A3 6A 56  29 20 39 32
(V) 20 38 32 20  30 32 20 39  41 20 20 36  44 36 20 38
(V) 32 8A 02 A2  92 9A CA 28  56 29 20 33  36 20 41 38
(V) 20 34 20 42  34 20 02 82  92 0D 0A 1B  33 36 20 31
(E) Serial buffer overflow
V) 32 8A 02 A2  92 9A CA 28  56 29 20 33  36 20 41 38
(V) 20 34 20 42  34 20 02 82  92 0D 0A 1B  33 36 20 31
(E) Serial buffer overflow
(E) 42 20 35 42  20 02 02 92  82 B2 8A 02  20 45 6E 64
(E) 65 64 20 22  95 99 85 D5  B1 D1 63 61  73 65 20 C1
(E) Serial FIFO overflow
(E) A5 B9 68 B4  05 6A 6B CB  EB B9 64 61  56 A8 48 68
(E)
(E) Serial FIFO overflow
(E) Ended up in default case while unwrapping...(tag 28)
(W) Unknown data received
(V)   payload:
(V) 28 56 29 20  20 20 70 61  79 6C 6F 61  64 3A 0D 0D
(V) 0A 1B 5B 30  6D 28 56 29  20 30 42 20  30 32 20 32
(V) 30 20 33 33  20 20 33 36  20 32 30 20  33 33 20 33
(V) 31 20 20 32  30 20 32 30  20 33 30 20  33 32 20 20
(V) 32 30 20 33  30 20 34 34  20 32 30 20  0D 0D 0A 1B
(E) Serial buffer overflow
V) 31 20 20 32  30 20 32 30  20 33 30 20  33 32 20 20
(V) 32 30 20 33  30 20 34 34  20 32 30 20  0D 0D 0A 1B
(E) Serial buffer overflow
(E) 5B 30 6D 28  56 29 20 33  30 20 33 32  20 32 30 20
(E) 33 30 20 20  33 32 20 33  38 20 33 32  20 32 30 20
(E) Serial FIFO overflow
(E) 20 41 31 20  41 33 20 36  41 20 35 36  20 20 32 39
(E) 20 32 30 20  33 39 20 33  32 20 0D 0D  0A 1B 5B 30
(E) 6D 28 56 29  20 32 30 20  33 38 20 33  32 20 32 30
(E) Serial FIFO overflow
(E) 20 20 33 30  20 33 32 20  32 30 20 33  39 20 20 34
(E) Serial FIFO overflow
(E) 31 20 32 30  20 32 30 20  33 36 20 20  34 34 20 33
E) Serial FIFO overflow
(E) 31 20 32 30  20 32 30 20  33 36 20 20  34 34 20 33
(E) Serial FIFO overflow
(E) Serial FIFO overflow
E) Serial FIFO overflow
(E) 39 32 20 39  41 20 43 41  20 32 38 20  20 35 36 20
(E) 32 39 20 32  30 20 33 33  20 20 33 36  20 32 30 20
(E) Serial FIFO overflow
(E) 34 31 20 33  38 20 0D 0D  0A 34 20 36  31 20 20 35
(E) 36 20 41 38  20 34 38 20  36 38 20 0D  0D 0A 1B 5B
(E) Serial FIFO overflow
(E) 30 6D 28 45  29 20 0D 0D  0A 1B 5B 30  6D 28 45 29
(E) 20 53 65 72  69 61 6C 20  46 49 46 4F  20 6F 76 65
(E) Serial FIFO overflow
(E) 72 66 6C 6F  77 0D 0D 0A  1B 5B 30 6D  28 45 29 20
E) 20 53 65 72  69 61 6C 20  46 49 46 4F  20 6F 76 65
(E) Serial FIFO overflow
(E) 72 66 6C 6F  77 0D 0D 0A  1B 5B 30 6D  28 45 29 20
(E) Serial FIFO overflow
(E) 7Serial FIFO overflow
(E) 5 6C 74 20  63 61 73 65  20 77 68 69  6C 65 20 75
(E) Serial FIFO overflow
(E) 6E 77 72 61  70 70 69 6E  67 2E 2E 2E  28 74 61 67
(E) 20 32 38 29  0D 0D 0A 1B  5B 30 6D 28  57 29 20 55
(E) Serial FIFO overflow
(E) 6E 6B 6E 6F  77 6E 20 64  61 74 61 20  72 65 63 65
(E) 69 76 65 64  0D 0D 0A 1B  5B 30 6D
(E) Serial FIFO overflow
(D) Serving /data.json over http...
(E) Serial FIFO overflow
(E) Ended up in default case while unwrapping...(tag 28)
(W) Unknown data received
(V)   payload:
(V) 28 56 29 20  20 20 70 61  79 6C 6F 61  64 3A 0D 0D
(V) 0A 1B 5B 30  6D 28 56 29  20 32 38 20  35 36 20 32
(V) 39 20 32 30  20 20 32 30  20 32 30 20  37 30 20 36
(V) 31 20 20 37  39 20 36 43  20 36 46 20  36 31 20 20
(V) 36 34 20 33  41 20 30 44  20 30 44 20  0D 0D 0A 1B
(E) Serial buffer overflow
V) 31 20 20 37  39 20 36 43  20 36 46 20  36 31 20 20
(V) 36 34 20 33  41 20 30 44  20 30 44 20  0D 0D 0A 1B
(E) Serial buffer overflow
(E) 5B 30 6D 28  56 29 20 30  41 20 31 42  20 35 42 20
(E) 33 30 20 20  36 44 20 32  38 20 35 36  20 32 39 20
(E) Serial FIFO overflow
(E) 20 32 30 20  33 30 20 34  32 20 32 30  20 20 33 30
(E) 20 33 32 20  32 30 20 33  32 20 0D 0D  0A 1B 5B 30
(E) Serial FIFO overflow
(E) 6D 28 56 29  20 33 30 20  32 30 20 33  33 20 33 33
(E) 20 20 32 30  20 32 30 20  33 33 20 33  36 20 20 32
(E) Serial FIFO overflow
(E) 30 20 33 32  20 33 30 20  32 30 20 20  33 33 20 33
(E) 33 20 32 30  20 33 33 20  0D 0D 0A 1B  5B 30 6D 28
(E) Serial FIFO overflow
(E) 56 29 20 33  31 20 32 30  20 32 30 20  33 32 20 20
E) 33 20 32 30  20 33 33 20  0D 0D 0A 1B  5B 30 6D 28
(E) Serial FIFO overflow
(E) 56 29 20 33  31 20 32 30  20 32 30 20  33 32 20 20
(E) Serial FIFO overflow
(E) Serial FIFO overflow
E) Serial FIFO overflow
(E) 32 30 20 32  30 20 0D 0D  0A 6F 6E 20  6F 76 65 72
(E)  4(E)  45
E)  4(E) Serial FIFO overflow
(E) Serial FIFO overflow
E) 20 45 6E 64  65 64 20 75  70 20 69 6E  20 64 65 66
(E) 61 75 6C 74  20 63 Serial FIFO overflow
(E) 61 73  65 20 77 68  69 6C 65 20
(E) 75 6E 77 72  61 70 70 69  6E 67 2E 2E  2Serial FIFO overflow
E) 61 75 6C 74  20 63 Serial FIFO overflow
(E) 61 73  65 20 77 68  69 6C 65 20
(E) 75 6E 77 72  61 70 70 69  6E 67 2E 2E  2Serial FIFO overflow
(E) E 28 74 61
(E) 67 20 32 38  29 0D 0D 0A  1B 5B 30 6D  28 57 29 20
(E) 55 6E 6B 6E  6F 77 6E 20  64 61 74 61  20 72 65 63
(E) 65 69 76 65  64 0D 0D 0A  1B 5B 30 6D
(E) Serial FIFO overflow
(E) Ended up in default case while unwrapping...(tag 28)
(W) Unknown data received
(E) Serial buffer overflow
(I) Increasing RX buffer to 2048 bytes
(V)   payload:
(V) 28 56 29 20  20 20 70 61  79 6C 6F 61  64 3A 0D 0D
(V) 0A 1B 5B 30  6D 28 56 29  20 32 38 20  35 36 20 32
(V) 39 20 32 30  20 20 32 30  20 32 30 20  37 30 20 36
(V) 31 20 20 37  39 20 36 43  20 36 46 20  36 31 20 20
(V) 36 34 20 33  41 20 30 44  20 30 44 20  0D 0D 0A 1B
(E) Serial buffer overflow

gskjold, 2023-12-25:
Is this a development board? I would avoid UART0 and instead use UART2 for ESP32

effix1983, 2023-12-25:
[image]
im using a  nodemcu wemos esp32 d1 mini cp2104.
I will try changing GPIO.

effix1983, 2023-12-26:
I switched gpio with no success. backed up the config and replaced the esp32 board, and now everything works. Guess the board was faulty.

closed 2023-12-26 by effix1983