UtilitechAS/amsreader-firmware #712: NES 83335-X not recognized from DSMR message

Jawar19, 2024-01-11:
**Describe the bug**
The AMSLeser P1 does not correctly identify the meter maker, therefor leaving some webinterface options as unknown.

**To Reproduce**
Steps to reproduce the behavior:
Go to the web interface while connected to a NES 83335-X and look at the meter configuration. The make and ID are unknown.
The frontpage then only shows unknown under the power meter.

I had problems with a HAN: footer checksum error in the beginning, I don't know is this is related and today I have not observed this HAN error while writing this.

I am waiting for further documentation of the Danish implementation of the DSMR protocol from the utility company. I would like to forward this when I get it.

**Expected behavior**
I would expect the meter make to be available as the DSMR message contains the necessary info.

**Screenshots**
[image]
[image]

**Hardware information:**
 - Country: Denmark
 - Meter: NES-83335-X
 - Encryption enabled: NO
 - AMS reader: POW-P1
 - M-bus adapter (if applicable): NO

**Relevant firmware information:**
 - Version: Latest
 - MQTT: Yes
 - MQTT payload type: HomeAssistant
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: 115200 8E1
 - Temperature sensors: N/A
 - ENTSO-E API enabled: NO

**Additional context**
From logging over TELNET
converted:
    > /NES5:83335-3E
    > 
    > 0-0:1.0.0(20240111 09:35:00)
    > 1-0:1.8.0(00036.273*kWh)
    > 1-0:2.8.0(00000.000*kWh)
    > 1-0:3.8.0(00000.137*kVArh)
    > 1-0:4.8.0(00027.954*kVArh)
    > 1-0:1.7.0(0000.193*kW)
    > 1-0:2.7.0(0000.000*kW)
    > 1-0:3.7.0(0000.000*kVAr)
    > 1-0:4.7.0(0000.221*kVAr)
    > 1-0:21.7.0(0000.193*kW)
    > 1-0:22.7.0(0000.000*kW)
    > 1-0:41.7.0(0000.000*kW)
    > 1-0:42.7.0(0000.000*kW)
    > 1-0:61.7.0(0000.000*kW)
    > 1-0:62.7.0(0000.000*kW)
    > 1-0:23.7.0(0000.000*kVAr)
    > 1-0:24.7.0(0000.221*kVAr)
    > 1-0:43.7.0(0000.000*kVAr)
    > 1-0:44.7.0(0000.000*kVAr)
    > 1-0:63.7.0(0000.000*kVAr)
    > 1-0:64.7.0(0000.000*kVAr)
    > 1-0:32.7.0(237.2*V)
    > 1-0:52.7.0(238.7*V)
    > 1-0:72.7.0(238.5*V)
    > 1-0:31.7.0(001.3*A)
    > 1-0:51.7.0(000.0*A)
    > 1-0:71.7.0(000.0*A)
    > !F302

RAW:
`DSMR frame:
2F 4E 45 53  35 3A 38 33  33 33 35 2D  33 45 0D 0A
0D 0A 30 2D  30 3A 31 2E  30 2E 30 28  32 30 32 34
30 31 31 31  20 30 39 3A  33 35 3A 30  30 29 0D 0A
31 2D 30 3A  31 2E 38 2E  30 28 30 30  30 33 36 2E
32 37 33 2A  6B 57 68 29  0D 0A 31 2D  30 3A 32 2E
38 2E 30 28  30 30 30 30  30 2E 30 30  30 2A 6B 57
68 29 0D 0A  31 2D 30 3A  33 2E 38 2E  30 28 30 30
30 30 30 2E  31 33 37 2A  6B 56 41 72  68 29 0D 0A
31 2D 30 3A  34 2E 38 2E  30 28 30 30  30 32 37 2E
39 35 34 2A  6B 56 41 72  68 29 0D 0A  31 2D 30 3A
31 2E 37 2E  30 28 30 30  30 30 2E 31  39 33 2A 6B
57 29 0D 0A  31 2D 30 3A  32 2E 37 2E  30 28 30 30
30 30 2E 30  30 30 2A 6B  57 29 0D 0A  31 2D 30 3A
33 2E 37 2E  30 28 30 30  30 30 2E 30  30 30 2A 6B
56 41 72 29  0D 0A 31 2D  30 3A 34 2E  37 2E 30 28
30 30 30 30  2E 32 32 31  2A 6B 56 41  72 29 0D 0A
31 2D 30 3A  32 31 2E 37  2E 30 28 30  30 30 30 2E
31 39 33 2A  6B 57 29 0D  0A 31 2D 30  3A 32 32 2E
37 2E 30 28  30 30 30 30  2E 30 30 30  2A 6B 57 29
0D 0A 31 2D  30 3A 34 31  2E 37 2E 30  28 30 30 30
30 2E 30 30  30 2A 6B 57  29 0D 0A 31  2D 30 3A 34
32 2E 37 2E  30 28 30 30  30 30 2E 30  30 30 2A 6B
57 29 0D 0A  31 2D 30 3A  36 31 2E 37  2E 30 28 30
30 30 30 2E  30 30 30 2A  6B 57 29 0D  0A 31 2D 30
3A 36 32 2E  37 2E 30 28  30 30 30 30  2E 30 30 30
2A 6B 57 29  0D 0A 31 2D  30 3A 32 33  2E 37 2E 30
28 30 30 30  30 2E 30 30  30 2A 6B 56  41 72 29 0D
0A 31 2D 30  3A 32 34 2E  37 2E 30 28  30 30 30 30
2E 32 32 31  2A 6B 56 41  72 29 0D 0A  31 2D 30 3A
34 33 2E 37  2E 30 28 30  30 30 30 2E  30 30 30 2A
6B 56 41 72  29 0D 0A 31  2D 30 3A 34  34 2E 37 2E
30 28 30 30  30 30 2E 30  30 30 2A 6B  56 41 72 29
0D 0A 31 2D  30 3A 36 33  2E 37 2E 30  28 30 30 30
30 2E 30 30  30 2A 6B 56  41 72 29 0D  0A 31 2D 30
3A 36 34 2E  37 2E 30 28  30 30 30 30  2E 30 30 30
2A 6B 56 41  72 29 0D 0A  31 2D 30 3A  33 32 2E 37
2E 30 28 32  33 37 2E 32  2A 56 29 0D  0A 31 2D 30
3A 35 32 2E  37 2E 30 28  32 33 38 2E  37 2A 56 29
0D 0A 31 2D  30 3A 37 32  2E 37 2E 30  28 32 33 38
2E 35 2A 56  29 0D 0A 31  2D 30 3A 33  31 2E 37 2E
30 28 30 30  31 2E 33 2A  41 29 0D 0A  31 2D 30 3A
35 31 2E 37  2E 30 28 30  30 30 2E 30  2A 41 29 0D
0A 31 2D 30  3A 37 31 2E  37 2E 30 28  30 30 30 2E
30 2A 41 29  0D 0A 21 46  33 30 32 0D  0A`

ArnieO, 2024-01-11:
Thank you for input regarding meter model; we'll fix this in next firmware patch. 👍 

> I had problems with a HAN: footer checksum error in the beginning, I don't know is this is related and today I have not observed this HAN error while writing this.

Try to increase the buffer size in config / meter section. 
@gskjold : Maybe the default buffer size for Pow-P1 should be increased?

gskjold, 2024-01-11:
Agree, will add a higher default for p1

Jawar19, 2024-01-11:
> Thank you for input regarding meter model; we'll fix this in next firmware patch. 👍
>
>
> Try to increase the buffer size in config / meter section. @gskjold : Maybe the default buffer size for Pow-P1 [...]

I have already the buffer set at the max of 4096, when I lowered it to 2048 bytes I got the FIFO overflow error. That being said I don't think I have actually seen the issue after explicitly setting the buffer at 4096bytes. whereas before it where automatically set by the POW P1 to 4096 bytes, I don't know how the auto-allocation works.

Jerroder, 2024-02-02:
My Pow-P1 running https://github.com/UtilitechAS/amsreader-firmware/commit/8a59fcb89a2ce9e37b2b83b4893b0320aa4e09bb is connected to an Omnia e-meter and I also get the `HAN: Footer checksum error` every ~30 seconds. My buffer size is 4096 and I also tried to set it to 2048 and back to 4096 but it didn't solve it. Is it possible to increase it even more in newer version?

ArnieO, 2024-02-03:
> My Pow-P1 running [8a59fcb](https://github.com/UtilitechAS/amsreader-firmware/commit/8a59fcb89a2ce9e37b2b83b4893b0320aa4e09bb) is connected to an Omnia e-meter and I also get the `HAN: Footer checksum error` every ~30 seconds. My buffer size is 4096 and I also tried to set it to 2048 and back to 4096 but it didn't solve it. Is it possible to increase it even more in newer version?

While waiting for @gskjod to weigh in: If the issue is checksum error I so do not think increasing the buffer size will help. How long is your cable? 

Could your run a Telnet debug (https://www.amsleser.no/module/ets_blog/blog?id_post=24) and post the result here of one of those payloads with Footer checksum error?

Jerroder, 2024-02-03:
My cable is 30cm long, it's the one I got from Lectronz.

Here's the payload (I had multiple in a row when connected to telnet):
```
(D) Serving /data.json over http...
(E) Serial buffer overflow
(E) Serial FIFO overflow
(E) Serial buffer overflow
(V) DSMR frame:
(V) 2F 4B 41 4D  35 0D 0A 0D  0A 30 2D 30  3A 31 2E 30
(V) 2E 30 28 32  34 30 32 30  33 31 34 34  39 34 30 57
(V) 29 0D 0A 31  2D 30 3A 31  2E 38 2E 30  28 30 30 30
(V) 30 34 39 39  36 2E 39 30  39 2A 6B 57  68 29 0D 0A
(V) 31 2D 30 3A  32 2E 38 2E  30 28 30 30  30 30 30 30
(V) 30 30 2E 30  30 30 2A 6B  57 68 29 0D  0A 31 2D 30
(V) 3A 33 2E 38  2E 30 28 30  30 30 30 30  30 34 34 2E
(V) 38 31 33 2A  6B 56 41 72  68 29 0D 0A  31 2D 30 3A
(V) 34 2E 38 2E  30 28 30 30  30 30 30 34  38 36 2E 34
(V) 39 32 2A 6B  56 41 72 68  29 0D 0A 31  2D 30 3A 31
(V) 2E 37 2E 30  28 30 30 30  30 2E 37 31  34 2A 6B 57
(V) 29 0D 0A 31  2D 30 3A 32  2E 37 2E 30  28 30 30 30
(V) 30 2E 30 30  30 2A 6B 57  29 0D 0A 31  2D 30 3A 33
(V) 2E 37 2E 30  28 30 30 30  30 2E 30 30  30 2A 6B 56
(V) 41 72 29 0D  0A 31 2D 30  3A 34 2E 37  2E 30 28 30
(V) 30 30 30 2E  33 36 37 2A  6B 56 41 72  29 0D 0A 31
(V) 2D 30 3A 32  31 2E 37 2E  30 28 30 30  30 30 2E 32
(V) 34 37 2A 6B  57 29 0D 0A  31 2D 30 3A  34 31 2E 37
(V) 2E 30 28 30  30 30 30 2E  30 32 34 2A  6B 57 29 0D
(V) 0A 31 2D 30  3A 36 31 2E  37 2E 30 28  30 30 30 30
(V) 2E 34 34 33  2A 6B 57 29  0D 0A 31 2D  30 3A 32 32
(V) 2E 37 2E 30  28 30 30 30  30 2E 30 30  30 2A 6B 57
(V) 29 0D 0A 31  2D 30 3A 34  3A 32 34 2E  37 2E 30 28
(V) 30 30 30 30  2E 30 30 30  2A 6B 56 41  72 29 0D 0A
(V) 31 2D 30 3A  34 34 2E 37  2E 30 28 30  30 30 30 2E
(V) 31 38 32 2A  6B 56 41 72  29 0D 0A 31  2D 30 3A 36
(V) 34 2E 37 2E  30 28 30 30  30 30 2E 32  31 39 2A 6B
(V) 56 41 72 29  0D 0A 31 2D  30 3A 33 32  2E 37 2E 30
(V) 28 32 33 34  2E 39 2A 56  29 0D 0A 31  2D 30 3A 35
(V) 32 2E 37 2E  30 28 32 33  32 2E 37 2A  56 29 0D 0A
(V) 31 2D 30 3A  37 32 2E 37  2E 30 28 32  33 34 2E 36
(V) 2A 56 29 0D  0A 31 2D 30  3A 33 31 2E  37 2E 30 28
(V) 30 30 31 2E  33 2A 41 29  0D 0A 31 2D  30 3A 35 31
(V) 2E 37 2E 30  28 30 30 30  2E 38 2A 41  29 0D 0A 31
(V) 2D 30 3A 37  31 2E 37 2E  30 28 30 30  32 2E 31 2A
(V) 41 29 0D 0A  21 38 41 45  46 0D 0A
(W) Frame checksum error (7aeb)
(D) Serving /data.json over http...
```

So I get a FIFO overflow even with 4096 as buffer size. I guess it doesn't appear in the web UI because it doesn't refresh quickly enough.

ArnieO, 2024-02-03:
OK, interesting. That reveals several tings, and could indeed be a buffer issue. 
We need @gskjold to analyze. 

How often does your meter send data? It varies a bit, we do not have full overview. Is it 1 second interval on the Omnia E-meter?

Jerroder, 2024-02-03:
Seems to be every ~10 seconds, according to how often the values refreshe in the web UI and the history of data in HomeAssistant.

ArnieO, 2024-02-03:
Sorry I should have said: Watch the LED on the device to see the interval. 

There is a very short blue blink (115 200 baud is fast...) generated by the arriving data, followed by a green blink in case payload is understood (but not necessarily accepted; parity checked etc).

Jerroder, 2024-02-05:
The LED blinks blue/green every 10 seconds also

ArnieO, 2024-02-05:
> The LED blinks blue/green every 10 seconds also

OK, thank you!
Interesting case that we need to understand. P1 meters have a larger payload than most other meters, but we expected the currently largest buffer size to be sufficient. 

We'll analyze your input and get back later this week.

ArnieO, 2024-02-08:
> (I had multiple in a row when connected to telnet)
Could be an interesting indicator. 
@gskjold will follow-up of this issue from here.

gskjold, 2024-02-08:
There is a possibility that debugging slows down the firmware enough to experience buffer overflow. If debugging is continuously activated on your device, I suggest trying to switch it off to see if it becomes stable.

Jerroder, 2024-02-08:
I definitely see the error more often when it's enabled, but it still happens when disabled, and it is always disabled.

gskjold, 2024-02-08:
OK, thanks, that is interesting. Since the payload is 571 bytes, I see no reason for buffer overflow with 4096 byte buffer... Maybe there is something else that takes up processing time and holds up the program, but not sure what that could be. Have you tried running it without MQTT enabled? Just trying to narrow down the cause

Jerroder, 2024-02-08:
I don't see the error in the UI anymore. In HomeAssistant's history I can see that a some measurements are missing (as they're supposed to come every 10 seconds, I can see that sometimes it jumps one and get a measurement 20 seconds later instead) and the UI doesn't refresh but no checksum error appear anymore.

I'm wondering if it's related to uptime? it's slower after boot but manages to "catch up" after some time? I'll monitor it keep you updated

closed 2024-03-21 by gskjold