UtilitechAS/amsreader-firmware #716: Expose tariff thresholds to MQTT

slippern1, 2024-01-16:
Is it possible to expose the configured tariff thresholds to MQTT?

ArnieO, 2024-01-18:
Yes, technically possible - but it is a bit difficult to understand the need, as these values are static. If you need them in Home Assistance (for instance), you can just enter them there. 

What is the intended way to use that information, and why would it be useful to receive them over MQTT?

slippern1, 2024-01-22:
I want to use the tariff thresholds in node-red flows without programming them manually.
My grid company recently changed the thresholds from a few to 20 steps.

gskjold, 2024-06-06:
Here is a version with this feature:
[esp32s3.zip](https://github.com/user-attachments/files/15685394/esp32s3.zip)
[esp32solo.zip](https://github.com/user-attachments/files/15685395/esp32solo.zip)
[esp32.zip](https://github.com/user-attachments/files/15685396/esp32.zip)
[esp32c3.zip](https://github.com/user-attachments/files/15685400/esp32c3.zip)
[esp8266.zip](https://github.com/user-attachments/files/15685401/esp8266.zip)
[esp32s2.zip](https://github.com/user-attachments/files/15685402/esp32s2.zip)

slippern1, 2024-06-09:
Thanks, i will test within this week!

slippern1, 2024-06-12:
It sure does work :-)
From an automation / developer view, an array or something with the values in the same entity is probably easier to work with. 

[image]

closed 2024-06-15 by gskjold