UtilitechAS/amsreader-firmware #718: Question: Sagemcom T210-D-r "Energienetze Steiermark"

cocoon77, 2024-01-19:
Hi,
could you please check if the Sagemcom T210-D-r also work with your Software.

**Hardware information:**
 - Country: Austria
 - Meter:  Sagemcom T210-D-r
 - Encryption enabled [yes]
 - AMS reader: don't have one. I have to check which one I need :/

Here is a link for more informations about the device (sorry in german)
[https://github.com/debug-richard/sagemcom-dsmr](https://github.com/debug-richard/sagemcom-dsmr)

thx for all your great work
Martin

ArnieO, 2024-01-19:
The software is not the issue here, the important point is to select the right hardware. 

- Your first link says it transmits at 115 200 baud, meaning it is "P1 companion standard". 
- Your second link, from e-netze.at, also says that their Sagemcom model is "P1 companion standard".

So according to this, the Pow-P1 should be right for you.

(And I now suspect we have an error in this table: https://github.com/UtilitechAS/amsreader-firmware/wiki/Known-hardware-configurations )

Noschvie, 2024-01-19:
https://www.e-netze.at/downloads-data/pdf.aspx?pdf=20240111_EN_Update%20Kundenschnittstelle%20Smart%20Meter_ID3884%20(01_2024)_V1.pdf

Have you had a look to this document?

cocoon77, 2024-01-19:
> https://www.e-netze.at/downloads-data/pdf.aspx?pdf=20240111_EN_Update%20Kundenschnittstelle%20Smart%20Meter_ID3884%20(01_2024)_V1.pdf
> 
> Have you had a look to this document?

Yes thats the 2nd link in my post :)

closed 2024-01-20 by cocoon77