UtilitechAS/amsreader-firmware #720: [v2.2.21] CGM decryption fails, if auth tag is present in the encrypted DLMS APDU, but no auth key is available

GEAG, 2024-01-20:
**Describe the bug**

It has been found, that some DSO in Switzerland with L&G E450 SM do not only _encrypt_ the DLMS APDU but also deliver the _authentication tag_ appended to the encrypted APDU, while at the same time, do not provide to the customer the _authentication key_. 

According to the encryption specification, authentication of the data issuer by an _authentication tag_ is an optional security feature. What we have found very special is, that the DSO did not provide the customer the _authentication key_, even after the customer has asked the DSO multiple times for it. But the DSO delivers the _authentication tag_ appended to the encrypted DLMS APDU with every telegram.

Our general recommendation for any DSO would be to:
1. either provide the _authentication key_ at least upon request by the customer if the DSO has append an _authentication tag_
2. or, if they have decided, to only provide an _encryption key_ to the customer, to not append the _authentication tag_ at all

In both these case, GcmParser.cpp has been found to decrypt the encrypted DLMS APDU with no problem. 

But in the special case that:
1. the encrypted DLMS APDU is delivered by the SM CII (configured by the DSO) includes the appended _authentication tag_
2. and the DSO does not provide the corresponding authentication key to its customer, 
the code in GcmParser.cpp ov v2.2.21 fails do decrypt that encrypted DLMS APDU and only tells "decryption failed" in the debug log.

In that combination when the DLMS APDU is encrypted as well as the _authentication tag_ appended to it, the _security byte_ at pos 13.dec in the header of the encrypted DLMS APDU (starts with 0xDB at pos 0.dec) is set to **0x30**.

The code for the decryption and authentication process

**To Reproduce**
Steps to reproduce the behavior:
1. Must have a DSO that has configured its SM to encrypt the data and append the authentication tag (_security byte_ = 0x30).
3. Do not configure the _authentication key_ in through the webUI (just the _encryption key_)
4. Check the debug log

**Expected behavior**
The decryption should also succeed, even if no _authentication key_ is configured, just the correct _decryption key_.

**Screenshots**
irrelevant

**Hardware information:**
 - Country: [Switzerland]
 - Meter: [Landis+Gyr E450]
 - Encryption enabled [yes]
 - AMS reader: [ESP-C3, own HW]
 - M-bus adapter (irrelevant):

**Relevant firmware information:**
 - Version: [2.2.21]
 - MQTT: [irrelevant]
 - MQTT payload type: [irrelevant]
 - HAN GPIO: [irrelevant]
 - HAN baud and parity: [2400 8E1]
 - Temperature sensors [irrelevant]
 - ENTSO-E API enabled: [irrelevant]

**Additional context**

A reworked and tested source code of GcmParser.cpp v2.2.21 will be provided by GEAG under the same AGPL license as the original v2.2.21 code. Please consult the file appended to this issue.

ArnieO, 2024-01-21:
Thank you for the heads up and code, we will look into this.

gskjold, 2024-01-26:
Thank you for the thorough feedback. As I read it, the following commit is sufficient to remedy this specific situation: https://github.com/UtilitechAS/amsreader-firmware/commit/e575ab755c540cec450b49ff1ba0e55bc2cbbea9

closed 2024-03-21 by gskjold