UtilitechAS/amsreader-firmware #722: MQTT Message LWT online/offline appreciated

tremichl, 2024-01-25:
Hi!

Not a problem but:

A MQTT Message LWT online/offline would be very helpful somtimes. Also presenting the actual IP-address via MQTT would be fine. 

Thanks for sharing your nice project.

Regards, Michael

ArnieO, 2024-01-26:
Thank you for the propositions. 
What is "LWT online/offline"?

tremichl, 2024-01-26:
LWT Last Will Testament is a MQTT feature. Explanation here: [https://cedalo.com/blog/mqtt-last-will-explained-and-example/](url)

Noschvie, 2024-01-26:
Tasmota documentation:

LWT Topic (Last Will and Testament)

On connection to the MQTT broker tasmota uses the Last Will and Testament (LWT) feature that instructs the broker to generate a notification about an ungracefully disconnect or timeout.

The topic to listen to for the state of this specific client is tele/<TOPIC>/LWT that with the mosquitto MQTT broker produces an output like:

```
$ mosquitto_sub -t "tele/tasmota_XXXXXX/LWT"
Offline
Online

```
The full LWT topic can be found in the tasmota console at boot:

`15:51:51.281 MQT: tele/tasmota_XXXXXX/LWT = Online (retained)`

gskjold, 2024-01-26:
Will be included in v2.3

closed 2024-02-15 by gskjold