UtilitechAS/amsreader-firmware #726: FW upgrade error -106

JuergStreit, 2024-02-06:
**Describe the bug**
FW uppgrade from v.2.2.21 to v2.2.18 fails with following error message:
Previous upgrade attempt (v2.2.28) does not match current version (v2.2.21) [0/**-106**]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'status'
2. Click on 'Install this version' (v2.2.28 is mentioned as "Latest version"
3. See error '-106'

**Expected behavior**
Upgrade to the new version

**Screenshots**
[image]

**Hardware information:**
 - Country: [Switzerland]
 - Meter: [L&G E570]
 - Encryption enabled [no]
 - AMS reader: [[ESP32-C3-DevKitM-1](http://10.0.20.21/vendor)]
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: [2.2.21]
 - MQTT: [no]
 - MQTT payload type: [-]
 - HAN GPIO: [GPIO7]
 - HAN baud and parity: [2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [no]

**Additional context**
It's my first gplug Adapter and is working so far; but I have to set the counter multipliers, which does not yet work in v2.2.21.

ArnieO, 2024-02-06:
Thank you for the feedback. 
I notice your Wi-Fi signal strength is quite low: RSSI at -85 dBm.
It will work fairly well for normal operation, but could lead to issues for an upgrading.

Can you move your device closer to your router/access point and you retry the upgrade?

JuergStreit, 2024-02-06:
@ArnieO thank you very, very much! 
Exactly this was the issue. Moving some centimeters, then with RSSI at -72dBm the upgrade worked like a charm.

closed 2024-02-06 by JuergStreit

ArnieO, 2024-02-06:
> @ArnieO thank you very, very much! Exactly this was the issue. Moving some centimeters, then with RSSI at -72dBm the upgrade worked like a charm.

Good to hear! 👍 

In next update of the user manual for Pow-U and Pow-P1 we will include something about positioning the unit and what the signal strength level should be in order to reduce the risk of failing the upgrade process. 

Disconnecting the device from the meter and bring it close to an access point (router of mesh node) is a good way to ensure good signal.

haldi4803, 2024-11-16:
```
Country: [Switzerland]
    Meter: [L&G E570]
    Encryption enabled [no]
    M-bus adapter (if applicable):
```

Thank you for that.
Just got an update to SmartMeter by EBL and it seems like they haven't finished the web interface for customers...
Time to do it ourselves.
M-Bus Adapter it is. Good to know.
This device is not yet in the known hardware configuration page. 

https://www.landisgyr.ch/product/landisgyr-e570/

ArnieO, 2024-11-17:
> This device is not yet in the known hardware configuration page.

You're right - thank you for noting that!
We will include this model (E570) in the [Selection guide](https://www.amsleser.no/index.php?fc=module&module=ams_selector&controller=display).