UtilitechAS/amsreader-firmware #728: Use Riemann sum integration with midpoint rule instead of right

stigvi, 2024-02-07:
**Is your feature request related to a problem? Please describe.**
No, not a huge problem, but something that makes it better

**Describe the solution you'd like**
I see from the sourcecode (amsdata.cpp, line 11) that you have implemented integration with something called right rule in the wikipedia article. This article also mention that midpoint rule is generally more accurate.

I have tested this in Home Assistant and used the Riemann sum integration to generate consumed energy from the power reported each 2.5 seconds. And this is compared with the energy used which is reported each hour and the energy used reported by amsleser on {publish topic}/realtime/import/hour. Using midpoint rule gives an error which is an order of magnitude lower than what the {publish topic}/realtime/import/hour has.

Implementing the midpoint rule is simple. Just take the average of the last and the second last active import power instead of using only last active import power.

_as such the middle Riemann sum is the most accurate approach to the Riemann sum_
https://en.wikipedia.org/wiki/Riemann_sum

_This method is the most accurate of the currently implemented methods, if the source updates often, since it better fits the curve of the intrinsic source._
https://www.home-assistant.io/integrations/integration/

Implementation in Home Assistant is on line 346
https://github.com/home-assistant/core/blob/dev/homeassistant/components/integration/sensor.py

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

SVH-Powel, 2024-02-08:
> Thank you for a great proposition for an improvement of the calculation of that integral! 👍

But I made an error when I wrote the feature request. All references to midpoint rule should be replaced to trapezoidal rule.

SVH-Powel, 2024-02-08:
But as I wrote, the implementation is easy. The average between the last and the second last is still true.

_For the trapezoidal rule, the function is approximated by the average of its values at the left and right endpoints of the subintervals._
From the wikipedia article ( https://en.wikipedia.org/wiki/Riemann_sum )

Ka8oum, 2024-02-21:
this would indeed be a great improvement for those energy estimations!

ArnieO, 2024-02-21:
Implemented in firmware rev2.3, which we expect to release in March.