UtilitechAS/amsreader-firmware #731: Bug or what... MY AMS do miss

LA5HJA, 2024-02-15:
**To Reproduce**
Like this all the time here.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware information:**
 - Country: Norway
 - Meter: Aidon Model 6525
 - Encryption enabled [yes/no]
 - AMS reader: [e.g. Pow-U, ESP32 etc]
 - M-bus adapter (if applicable):
[image]

**Relevant firmware information:**
 - Version: 2.2.28
 - MQTT: yes
 - MQTT Raw
 - HAN GPIO: [15
[image]

 - HAN baud and parity: [e.g. 2400 8E1]

 - ENTSO-E API enabled: yes

**Additional context**
Add any other context about the problem here.
Running 2 x ESP32 on the RS484 bus, both do miss Apmerage on L2, both run same SW and HW
[image]

tronde-ams, 2024-02-15:
Aidon 6525 does not give us the value for I2. It is impossible to calculate the value with the data available from the meter. Earlier FW tried to calculate it, but gave mostly useless values.

tronde-ams, 2024-02-15:
I wrote something in Norwegian here
https://github.com/UtilitechAS/amsreader-firmware/issues/686

ArnieO, 2024-02-16:
https://www.amsleser.no/module/ets_blog/blog?id_post=31

closed 2024-02-16 by ArnieO