UtilitechAS/amsreader-firmware #735: 802.11v and 802.11k support

SVH-Powel, 2024-02-21:
**Describe your problem**

This is a question.

If you are using the esp-idf library, have you enabled support for 802.11v BSS Transition Management and 802.11k Radio Resource Management?

Reason for asking is that I see that my Pow-U is connected to a point in the wifi mesh which is far from optimal.

ArnieO, 2024-02-21:
We use the latest available Arduino Framework for the code, which means the very latest feature in ESP-IDF may not be there yet. However, connecting to best BSSID (not just a random SSID node) is in the current framework, so it should connect to the best node in a mesh network and stay there.

SVH-Powel, 2024-02-21:
Well, mine connects to a nearby point when pow-u is restarted. The signal strength is then usually between -72 to -75 dbm.

But after a day or two, it always end up connecting to another point in the mesh and which have a signal strength between -85 to -90 dbm. And when connected there, it stays there for a week or two until it suddenly reboots. Probably because of weak signal.

And when it boots, it starts over again with connecting to the strongest point and a day later switching over to a weak point.

ArnieO, 2024-02-21:
OK, I see. I remember having seen the same issue here, but I got the impression that the network in some way handled that better after some time.

In my TPLink Deco mesh network I can in the app indicate preferred node for wireless clients. Have you tried that?

SVH-Powel, 2024-02-21:
> In my TPLink Deco mesh network I can in the app indicate preferred node for wireless clients. Have you tried that?

I am using Google Wifi (unfortunately) and there is no option for setting preferred node.

I see that the node it is switching to after a day or two, is the "master" in the mesh. It is the node which is connected to internet. Maybe that is automatically preferred in the Google wifi system. The other nodes are connected via a network cable.

ArnieO, 2024-02-21:
> there is no option for setting preferred node.

Is there an option for declaring a node as mobile (or not)? 
I think I had that in an earlier version of my system, but cannot find it now. Maybe disappeared with an update of the app.

gskjold, 2024-03-21:
Note to self: https://github.com/espressif/esp-idf/blob/2a4923faa695ba8b4e98bef7b7c7b7b7c1156ebd/components/esp_wifi/include/esp_wifi_types.h#L285

closed 2024-03-21 by gskjold