UtilitechAS/amsreader-firmware #737: Login with URL including username and password

sm1oii, 2024-02-24:
Hi

I have my pow-p1 reader and other devices under subdomains. I would like to log into my pow-p1 reader like this : 

https://username:password@p1.mydomain.com

This is a standard format that works exellent on my IP-cameras and other devices but NOT on pow-p1 reader :-(
Can this feature be included in next release?

/Harri

Noschvie, 2024-02-24:
Why not using a reverse proxy for that? Basic Auth and https

ArnieO, 2024-02-25:
> Why not using a reverse proxy for that? Basic Auth and https

Can you elaborate on how to do that?

sm1oii, 2024-02-25:
> Why not using a reverse proxy for that? Basic Auth and https

Sounds like a good idea but against company network policy and network restrictions.

Noschvie, 2024-02-25:
Have a look for example to pfSense, HAProxy and basic auth
[User Authentication with HAProxy on pfSense](https://hochwald.net/user-authentication-with-haproxy-on-pfsense/)[](https://hochwald.net/user-authentication-with-haproxy-on-pfsense/)
But also supported by Traefik and NGINX

gskjold, 2024-03-10:
Logging in with http://user:pass@mylocalip works fine, this must be related to your  proxy settings

closed 2024-03-10 by gskjold