UtilitechAS/amsreader-firmware #740: Landis+Gyr E450 as provisioned by Regionalwerke Baden: Reactive energy and phase currents

avandorp, 2024-02-29:
**Is your feature request related to a problem? Please describe.**
For the Landis+Gyr E450 as provisioned by Regionalwerke Baden Switzerland, AMS reader already shows import and export power, thanks! Reactive energy and phase current values are missing, but should be reported by the meter.

**Describe the solution you'd like**
If reactive power and phase current values could be parsed that would be tremendous.

**Additional context**
The power utility provides more info regarding the OBIS codes at https://www.regionalwerke.ch/fileadmin/dok/services/RWB_Flyer_SmartMeter_Bedienungsanleitung.pdf. 
I've translated the important parts:

```
OBIS Code; Attribute index	Register
Transmission interval 05s
0-8:25.9.0; 2	Object list push settings end user information 1
0-8:25.9.0; 1	OBIS Code  Push Settings end user information 1
0-0:96.1.0; 2	Device identification 1 (Producer serial number)
1-0:1.7.0; 2	Power import +P
1-0:2.7.0; 2	Power export -P
1-1:1.8.0; 2	Energy import +A (QI + QIV)
1-1:2.8.0; 2	Energy export  -A (QII + QIII)
1-1:5.8.0; 2	Reactive energy +Ri (QI)
1-1:6.8.0; 2	Reactive energy +Rc (QII)
1-1:7.8.0; 2	Reactive energy -Ri (QIII)
1-1:8.8.0; 2	Reactive energy -Rc (QIV)
1-0:31.7.0; 2	Current L1
1-0:51.7.0; 2	Current L2
1-0:71.7.0; 2	Current L3
Transmission interval 60s
0-9:25.9.0; 2	Object list push settings end user information 2
0-9:25.9.0; 1	OBIS Code  Push Settings end user information 2
1-1:1.8.1; 2	Energy import +A (QI + QIV) Tariff 1
1-1:1.8.2; 2	Energy import +A (QI + QIV) Tariff 2
1-1:2.8.1; 2	Energy export -A (QII + QIII) Tariff 1
1-1:2.8.2; 2	Energy export -A (QII + QIII) Tariff 2
Transmission interval 15min
(Other M-Bus values if wireless M-Bus capable devices are read by the meter)
```
I've added a file with a few frames from my reader. Please tell me, if you need more or different data.
[RegionalwerkeBadenSwitzerlandFrames.txt](https://github.com/UtilitechAS/amsreader-firmware/files/14454178/RegionalwerkeBadenSwitzerlandFrames.txt)

ArnieO, 2024-03-01:
Thank you for precise report and documentation, we will look into this.

ArnieO, 2024-03-14:
This format will be supported in v2.3 - which should soon be released. 
Test version is available here: https://github.com/UtilitechAS/amsreader-firmware/discussions/738
(Test version may lack some parameters.)

gskjold, 2024-03-14:
One thing I suddenly notice here, this meter is sending OBIS 5.8.0 through 8.8.0 which are capacitive and inductive reactive energy, how does this relate to 3.8.0 and 4.8.0 reactive energy? Can it be converted? We do not have any internal registers for capacitive and inductive energy.

avandorp, 2024-03-14:
They can be added. The question is: Which ones? As I'm not an electrical
engineer, I don't know those notations. I'll ask my power utility.

ArnieO, 2024-03-14:
> One thing I suddenly notice here, this meter is sending OBIS 5.8.0 through 8.8.0 which are capacitive and inductive reactive energy, how does this relate to 3.8.0 and 4.8.0 reactive energy? Can it be converted? We do not have any internal registers for capacitive and inductive energy.

I think the reactive parameters are of very little value for most of our users, except large industrial users that could be charged extra for reactive energy - as the grid has to be dimensioned for it. (Reactive power/energy is sometimes compared to[ the foam on top of the beer in a glass](https://blog.se.com/energy-management-energy-efficiency/2018/01/22/can-learn-power-pint-beer/): The glass must be large enough to handle the foam, but the user is really mostly interested in what is under the foam. 😆)

To understand this, we need to return to "basics":
Power/energy in the real world has an active and a reactive component. This can be illustrated by vectors in a current/voltage diagram. 
- S = Apparent power/energy = Length of the resulting vector, that can be decomposed into:
- P = Active power/energy: kW/kWh
- Q = Reactive power/energy: kVar/kVarh

Then:
- OBIS code 3.8.0 is the reactive part Q of the vector S. Q can either be inductive or capacitive, (in 1st or 2nd quadrant), in which case this meter reports it in either 5.8.0 or 6.8.0.
- OBIS code 4.8.0 is the same for exported energy: 3rd or 4th quadrant. 
So one of the values 5.8.0 or 6.8.0 is always zero, and then same for 7.8.0 vs 8.8.0. 

In other words: This meter gives the added information whether the reactive power P is inductive or capacitive, while meters that report OBIS codes 3.8.0/4.8.0 only gives the amount of reactive energy, disregarding whether it is inductive or capacitive (unless they also report phase angle; see below). 

This means that for this meter: `3.8.0 = 5.8.0 + 6.8.0` and `4.8.0 = 7.8.0 + 8.8.0`

In this illustration, current is the horizontal axis, voltage is the vertical axis. 
[image]
(Illustration from https://www.securemeters.com/uk/white_papers/import-and-export-of-electrical-energy/ )

We received the below information by email from Netz Burgenland some time ago. It is somewhat (but not quite) on the same topic, at least it illustrates the quadrants. Those meters report phase angle for the reactive energy. The phase angle tells whether the reactive energy is inductive or capacitive.

[image]

ArnieO, 2024-03-14:
To conclude: 
My proposition is to sum the parameters into our two "Reactive energy" registers (import and export).

gskjold, 2024-03-14:
Thanks, just what I needed!

avandorp, 2024-03-14:
> I think the reactive parameters are of very little value for most of our users, except large industrial users that could be charged extra for reactive energy - as the grid has to be dimensioned for it.

Ah, but the value can simply be educative, or in satisfying curiosity, or identifying something that's unexpected or even dangerous. I've learned a lot by simply looking at incoming data.

Looking forward to v2.3. Thanks for all you work.

closed 2024-03-15 by gskjold

gskjold, 2024-03-15:
Latest build in #738 has current sensors added

avandorp, 2024-03-15:
Firmware upload seems to work, but after "Device is upgrading, please wait" it still says v2.2.28 without voltage and amperage.

gskjold, 2024-03-16:
I am not sure what hardware you have, might be running out of power. Also, are you sure you are using correct bin file?

avandorp, 2024-03-16:
No luck. I've dismantled my installation and flashed via USB. But with the new firmware the device doesn't come up (esp32c3), no LEDs blinking or anything. Reflashing with 2.2.28 works, but I can't get the new firmware to do anything. Tried both an upgrade and a full flash with partitions and everything.