UtilitechAS/amsreader-firmware #744: Publish to Home assistant using HomeAssistant payload seems unstable

sorenbache, 2024-03-08:
**Describe your problem**
I have been running the AMS reader for 1 1/2 years now with an ESP 32 and Kamstrup - very stable and fine. And using the Home Assistant discovery payload. Yesterday, I ran into the issue that the ESP 32 suddenly disappeared. After some back and forth I realized that is was a faulty cable that led to unstable wifi. Before realizing that I erased and reflashed the ESP32 so I had to do a fresh setup. But all working now - except a sketchy connection to HA via MQTT.

[image]

My trouble comes with the MQTT connection - if I set it up to use JSON data comes through fine. But I want to use the HomeAssistant payload. Before the trouble the payload was found under the name "elforbrug" and I obviously wanted to reuse that - but data was not transferred, only the config like this:

[image]

I managed when changing to elforbrug2 to temporarily get meter data to appear in HA, but after changing to something else than "elforbrug2" (which I felt was a bad name) it stopped working again. Again, "elfobrug2" also appears in HA, but trying e.g. "elforbrug3" is unsuccessful so I am quite puzzled on what is going on here.

[image]

So any input on how to debug this further?

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup
 - Encryption enabled yes
 - AMS reader: ESP32
 - 
**Relevant firmware information:**
 - Version: 2.2.27 (and I have also tried 2.2.28)
 - MQTT: Yes
 - MQTT payload type: HomeAssistant
 - HAN GPIO: GPIO 17
 - HAN baud and parity: 2400 8N1
 - ENTSO-E API enabled: No

sorenbache, 2024-03-08:
An update here: I lost faith into the ESP32. So I flashed a new one - worked immediately after setting it up. Well well, still wonder how everything seems "green" but it did not really work with the integration to mqtt in HA.

ArnieO, 2024-03-09:
@gskjold: Could this be an "edge case" linked to inadequate publishing the of HA sensors?

gskjold, 2024-03-10:
I am not sure if we are seeing the whole config here, but from the screenshots it appears the "Publish topic" is empty, this is a required field

sorenbache, 2024-03-10:
For what it is worth I am pretty sure I tried all kinds of possibilities, both having a Publish topic filled out and also without. Nothing helped.

closed 2024-03-21 by gskjold