UtilitechAS/amsreader-firmware #745: My Pow-U is rebooting

slippern1, 2024-03-09:
**Describe your problem**
My Pow-U has suddenly began to reboot itself quite a lot now.
The device is USB powered.

Latest reboot reason: Reason: Exception reset (2/2)

How can i debug this?

**Hardware information:**
 - Country: [e.g. Norway]
 Norway
 - Meter: [e.g. Aidon]
 Aidon
 - Encryption enabled [yes/no]
 No
 - AMS reader: [e.g. Pow-U, ESP32 etc]
 Pow-U ESP esp8266

**Relevant firmware information:**
 - Version: v2.2.28
 - MQTT: yes
 - MQTT payload type: HomeAssistant
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [e.g. 2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]

gskjold, 2024-03-10:
Exception cause 2 is "InstructionFetchErrorCause - Processor internal physical address or data error during instruction fetch" which indicates a hardware failure. Is this always the reason?

slippern1, 2024-03-10:
Not always, now i have this reason:
Last boot: 10. Mar 09:11
Reason: Software restart (4/0)

ArnieO, 2024-03-10:
This is an old Pow-U with ESP8266, we have not sold those since spring 2022. 
We might be starting to reach end-of-life for number of read/write cycles to the flash memory.

In such cases we are proud to do excellent customer service, and replace the board.
So @slippern1 : Please contact me on post@amsleser.no and identify yourself. 
If you are not the original buyer, please tell us who bought it so we can verify in the customer data base.

slippern1, 2024-03-10:
@ArnieO You've got mail :-)

closed 2024-03-10 by gskjold