UtilitechAS/amsreader-firmware #746: Combined import/export value in data.json/MQTT

jladefoged, 2024-03-10:
**Is your feature request related to a problem? Please describe.**
Grid import and grid export values are available in data.json, but software like OpenDTU-onBattery does only support one value and expect this value to be negative when exporting. The value is used with using the dynamic power limiter feature.

**Describe the solution you'd like**
Combined import/export value in data.json and perhaps MQTT as well.
Perhaps an "ie" value?

**Describe alternatives you've considered**
Requesting OpenDTU-onBattery to support import and export as seperate values.

**Additional context**
Link to OpenDTU-onBattery project:
https://github.com/helgeerbe/OpenDTU-OnBattery

ArnieO, 2024-03-10:
This sounds like a reasonable improvement that can be useful for some users - unless someone sees a potential pitfall here?

Noschvie, 2024-03-10:
Suggest to add an additional end point for the "OpenDTU-onBattery" feature based on current / other solutions.

gskjold, 2024-04-14:
I'd like some more info about this. Lets start with what opendtu is and what other integrations are like

jladefoged, 2024-04-18:
OpenDTU-onBattery will support separate import/export values in their next firmware version, but adding support for at negative consumption value in AMS Reader might still be a good idea for other future projects. If not the feature request may be closed.

closed 2024-06-06 by gskjold