UtilitechAS/amsreader-firmware #748: Need help: UI translation to Danish

ArnieO, 2024-03-22:
The soon-to-be-released firmware version 2.3 will have a feature to change user interface language. 

So far we have the following languages ready:
[image]

But we do not yet have a translation to Danish. 😐 
[image]

If you would like to help us with that, please go to https://localazy.com/p/amsleser and start translating!

Saturate, 2024-03-22:
I started doing some of the translations, hope others will help as well!

ArnieO, 2024-03-23:
> I started doing some of the translations, hope others will help as well!

Thank you for starting, it would be great to have this done before we release v2.3. There are in total a bit more than 300 terms to translate. Should not take more than 1/2 hour in total (took me around that time when I did the translation to French).
There are automatic translation suggestions - but they cannot always be used. On some of the terms there are screenshots attached to help out with showing the context.
[image]

ArnieO, 2024-03-26:
Danish translation is now ready. Thank you very much for your help, @Saturate ! 🌹

Saturate, 2024-03-27:
@ArnieO No problem :) actually came to see if I could help out with the web part.

closed 2024-04-01 by ArnieO