UtilitechAS/amsreader-firmware #750: HAN: Payload boundary flag missing

sthayduk, 2024-03-23:
I am currently trying to get your cool solution to work with an Austrian Smart Meter from EVN. I'm connecting the Smart Meter via an M-BUS Slave Click from MIKROE to an ESP32. In principle, the software is running, but unfortunately, I am not receiving any data and the error message **HAN: Payload boundary flag missing**.

**Hardware information:**
 - Country: Austria
 - Meter: Kaifa MA309M
 - Encryption enabled: yes
 - AMS reader: ESP32
 - M-bus adapter: MIKROE TSS721ADR

**Relevant firmware information:**
 - Version: 2.2.28
 - MQTT: no
 - HAN GPIO: GPIO34
 - HAN baud and parity: 2400 / 8N1

Any Ideas what must be changed to get correct data?

Noschvie, 2024-03-23:
Have v2.2.28 running connected to Kaifa MA309M from EVN...

Which buffer size do you have configured?

[image]

ArnieO, 2024-03-23:
Did you try inverting the data signal?
[image]

Are you are sure you're getting data out of your [M-bus interface board](https://www.mikroe.com/m-bus-slave-click)? Measure its Tx output, you should see something happening there at regular intervals. 

A connection diagram that includes your microcontroller would also help.

sthayduk, 2024-03-23:
@Noschvie I use the same settings as you.

[image]

@ArnieO if I invert the data signal the error changes to "HAN: Unknown data received, check meter config". I'm sure I get data because if nothing is received, I also get another error message. 

And thats how I put the things together ...

![1E84B2B0-1A50-4BD2-BD95-285D59A7555C_1_102_o](https://github.com/UtilitechAS/amsreader-firmware/assets/43806330/6985
[image]
5ecf-e436-497e-865e-135476a2cdf3)
[image]
[image]

gskjold, 2024-03-25:
Maybe a debug would reveal what data you are actually receiving: https://www.amsleser.no/blog/post/24-telnet-debug

ArnieO, 2024-03-25:
@sthayduk 

1. It looks like you are reading the data on Rx0 on your board. Please try reading on one of the GPIOs.
2. Run a Telnet debug and show us what you receive: https://www.amsleser.no/blog/post/24-telnet-debug

gskjold, 2024-03-25:
Actually, maybe this is hardware vs software serial issue. You could try using "RX2" on your board and set the HAN GPIO to UART2

sthayduk, 2024-03-25:
Unfortunately, I won't be home until Thursday this week! I'll take another look at it then. I found a solution from another repo that also works with my hardware config: https://github.com/DomiStyle/esphome-dlms-meter. However, I would prefer your solution because it seems to be much more flexible!

ElMyggo, 2024-03-28:
I have different meter (kamstrup) but i have tested with same m-bus card, and i have connect from TX and to UART2 (GPIO16) (and 3.3V/GND) with 2400 8N1 and working with my ESP32 Wroom module

sthayduk, 2024-03-28:
> Actually, maybe this is hardware vs software serial issue. You could try using "RX2" on your board and set the HAN GPIO to UART2

Great! That was the solution! Thank you very much!

closed 2024-03-28 by sthayduk