UtilitechAS/amsreader-firmware #752: Wrong DNS when DHCP is enabled

Bruners, 2024-03-26:
When DHCP is enabled the DNS value sometimes gets the wrong value, causing price fetching to fail. Took a while to figure this out since price fetching would work for a while and then randomly started to fail. 

This is the only device on the network that has this problem.

[image]

ArnieO, 2024-03-27:
Interesting observation, but there could be other reasons. 

1. Which setting do you use for price fetching (Entso-E or our server)?
2. I don't recognize that DNS you are using. Did you try one of the larger, public ones - like 1.1.1.1?

Bruners, 2024-03-27:
Still getting issues now even with static ip settings. I have set static dns to 1.1.1.1/8.8.8.8. The DHCP (vendor locked) supplies rest of the network with 192.168.10.1 + fe80::becf:4fff:fe35:513d%wlan0

[image]
[image]

The settings for price fetching

[image]

ArnieO, 2024-03-27:
OK, you're reading prices directly from Entso-E. 
We have experienced relatively frequent issues with getting response from their API, so we have established an intermediate server. If you delete the Entso-E API key (but leave "Enable price fetch" enabled), you will read from our server instead.

Bruners, 2024-03-27:
Will give that a try

gskjold, 2024-04-02:
This is a known issue, outside of our control. This is related to IPv6 DNS being applied as if it was IPv4, even if IPv6 is not enabled in firmware. See https://github.com/espressif/arduino-esp32/issues/8221

The firmware monitors the DNS and changes it back on interval. Also, IPv6 will be available in v2.3

closed 2024-04-04 by gskjold