UtilitechAS/amsreader-firmware #754: Weird price value today

spacecosmos, 2024-03-31:
**Describe the bug**
Not sure what happened, but I was looking at the price value for today's usage (from midnight to now 08:50) and it showed a negative value of approximately -640 NOK.

Looking at the future energy price during the night, it hasn't been _that_ cheap. I then thought of restarting the unit, and now it shows even higher!

**Expected behavior**
Calculate correct value depending on the price given.

**Screenshots**
[image]

[image]

**Hardware information:**
 - Country: Norway
 - Meter: Kamstrup
 - Chip: esp32s2 (160MHz)
 - Device: Pow-K+

**Relevant firmware information:**
 - Version: [e.g. 2.1.0]
 - MQTT: [yes/no]
 - MQTT payload type: [e.g. JSON]
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [e.g. 2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]

**Additional context**
Not sure if I want to try rebooting again, afraid it would keep increasing.

spacecosmos, 2024-03-31:
Update: Now it's down a bit again (?).

[Screenshot_20240331_092402_Opera](https://github.com/UtilitechAS/amsreader-firmware/assets/55928670/43c04226-f9b7-4362-9134-ebb45d7cb2ec)

ArnieO, 2024-03-31:
@gskjold : Maybe some weird DST bug?

gskjold, 2024-03-31:
I hate DST... Probably...

closed 2024-04-09 by spacecosmos