UtilitechAS/amsreader-firmware #759: 2.3 live plot not working, causing browser tab crashes and freezes

ThorAlex87, 2024-04-06:
**Describe the bug**
When loading the web UI the browser tab slows down, freezes, crashes or has other issues. This happens only when realtime plot is active, if it's turned off everything seems to work fine. This happens in both firefox and chrome on my pc, but firefox on my android tablet seems to work fine except the realtime plot being completely empty. Problem also happens in /configuration and /status, so might be related to something happening in the background and not the visual plot itself.

**To Reproduce**
Go to the web UI in firefox or chrome on PC while realtime plot is set to auto or enabled.

**Expected behavior**
Everything works regardless of browser and device

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware information:**
 - Country: Norway
 - Meter: Kaifa
 - Encryption enabled no
 - AMS reader: Pow-U+

**Relevant firmware information:**
 - Version: 2.3.0
 - MQTT: yes
 - MQTT payload type: Home-assistant
 - ENTSO-E API enabled: yes

**Additional context**
Could be something about my pc, but I don't have another to test with.

gskjold, 2024-04-06:
Interesting issue, I have yet to see this myself, but I'm guessing there is something wrong here, and we'll definitely find it

ThorAlex87, 2024-04-06:
> Interesting issue, I have yet to see this myself, but I'm guessing there is something wrong here, and we'll definitely find it

Let me know if there is anything I can do to help :)

gskjold, 2024-04-06:
If you could try this build, that would be great:
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/14893223/esp32s2.zip)

Guide for flashing here: https://github.com/UtilitechAS/amsreader-firmware/wiki/Flashing-via-web-GUI

If there is still an issue, I'd like to see a screeshot of the debugger in Chrome (Press F12), in the console it will show some debugging:
[image]

ThorAlex87, 2024-04-06:
Thanks! No crashing or other performance issues wit the new build. The graph is still not working right, but it's not completely blank as before. 

[image]

gskjold, 2024-04-06:
You have production?

ThorAlex87, 2024-04-06:
> You have production?

Yeah, 15kwp solar and the sun is out so the -13000w number is real. 

(BTW, should that not scale to 13kw like the other graphs?)

gskjold, 2024-04-06:
This was the problem, scaling on that graph was broken :) Patch coming up soon

closed 2024-04-06 by ThorAlex87