UtilitechAS/amsreader-firmware #760: Hangs during upgrade to v2.3.0

fnorlund, 2024-04-06:
**Describe the bug**
It has worked like a charm but this morning it crashed when trying to upgrade from v.2.2.28 to v2.3.0. Reinstalled v2.2.28 firmware.bin using esptool which was successful but it still don't start. 

**To Reproduce**
Click upgrade on the GUI which previously worked perfectly

**Expected behaviour**
How do I avoid losing my settings?

**Screenshots**
Blank screen

**Hardware information:**
 - Country: Sweden
 - Meter: Kaifa
 - Encryption enabled: no
 - AMS reader: ESP32

**Relevant firmware information:**
 - Version: 2.3.0
 - MQTT: yes
 - MQTT payload type: Home-assistant
 - ENTSO-E API enabled: yes

gskjold, 2024-04-07:
Have it jumped back to AP mode maybe?

ArnieO, 2024-04-07:
> How do I avoid losing my settings?

Do a backup of the settings before running the update - but that is probably too late for you now. 
[image]

When downgrading back from v2.3.x to v2.2.x the unit will start up in "stealth" AP mode (nothing visible on the LED). You will have to search for the Access point (AMS2MQTT) and connect to it, then reconfigure the device on http://192.168.4.1/

fnorlund, 2024-04-08:
Yes, it was too late to rescue settings. 
Couldn't connect to AP and after various attempts I had to erase the flash memory to be able to re-install v2.3.1, using esptool. Then AP popped up again. Filled initial configuration settings dialogue and clicked save which made it freeze again.

ArnieO, 2024-04-09:
> Yes, it was too late to rescue settings. Couldn't connect to AP and after various attempts I had to erase the flash memory to be able to re-install v2.3.1, using esptool. Then AP popped up again. Filled initial configuration settings dialogue and clicked save which made it freeze again with blank screen.

Weird. Which hardware is this?

gskjold, 2024-04-09:
If this is reproducible, I need a step by step description on how to reproduce

fnorlund, 2024-04-09:
Thanks for your support. 

Yes, I understand and will provide a step-by-step description within the next few days. I've much on my table at the moment. 

This is the command I used to load v2.3.1. Is it correct?
$esptool --chip esp32 --port COM4 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader.bin 0x8000 partitions.bin 0x10000 firmware.bin

fnorlund, 2024-04-09:
The HW is Wemos LOLIN D32 with chip, according to esptool, ESP32-DOWD-V3

gskjold, 2024-04-09:
Should be sufficient, although here is what I use:
```
esptool --chip esp32 --port COM1 --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio -flash_freq 40m --flash_size detect 0x1000 bootloader.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 firmware.bin
```

fnorlund, 2024-04-20:
I've not yet tried to reproduce the error.

I just now thought of a possible reason for the crash. I probably forgot to include the file boot_app0.bin when I made the very first installation with v2.2.24 ish. Is it possible to get it working even if that file is missing?
If so, can this be the reason why the upgrade from 2.2.28 to 2.3.0 failed despite every upgrade from 2.2.24 to 2.2.28 was successful?

gskjold, 2024-04-21:
boot_app0.bin is needed to do OTA upgrades, but as far as I can tell you don't technically need it when flashing via USB

ArnieO, 2024-04-22:
I recommend to follow the procedure described in the new blogpost "[Updating device over USB cable, simplified method](https://www.amsleser.no/module/ets_blog/blog?id_post=34)"

fnorlund, 2024-04-24:
I've now upgraded two different devices from 2.2.28 to 2.3.5 and both were successful. For that reason I close the issue since I can't reproduce the crash.

Thanks for your support

closed 2024-04-24 by fnorlund