UtilitechAS/amsreader-firmware #763: Update of esp8266 Pow-K from 2.2.28 to 2.2.31

Hawkeye4060, 2024-04-07:
**Describe the bug**
Last evening I made an update as there was announced a new version (v2.2.31)
After the update and reboot, it seems that the Amsreader wasn't able to connect to the WiFi.
Today I have struggled to get it back to v2.2.28
I was able to reset and get back in AP-mode where I could upload the older firmware.
Now everything works again, but with v2.2.28.

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup
 - Encryption enabled: yes
 - AMS reader: Pow-K, ESP8266
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.2.28 been working since Dec.2023
 - MQTT: yes
 - MQTT payload type: JSON
 - HAN GPIO: GPIO12
 - HAN baud and parity: 2400 8E1
 - Temperature sensors: none
 

**Additional context**
After I have got it working again, it says that there is a new firmware, but I am not happy about to try again until I am sure that tis issue have been solved!?!?!?!

Best regards 
Mogens Hartmann

gskjold, 2024-04-07:
I'm trying to reproduce this one, but I am not able to... I have a ESP8266 Pow-K here, fully set up, and upgrade works without issues..

Hawkeye4060, 2024-04-07:
It would not boot or connect to Wi-Fi afterwards.
So you think I should try again? 😉

ArnieO, 2024-04-08:
The upgrade from v2.2.28 to v2.3.x was tested also on devices with ESP8266 before release - so this is a bit strange. It should normally go quite smoothly. 

Could it be that the Pow-K restarted in AP mode - for some reason? 
It is easier to identify that situation on devices with ESP8266 module, as there is a blue LED on the ESP-module that blinks while it is in AP mode (@gskjold : I assume that is still the case on v2.3.x, even if the yellow light is now steady?). Devices with ESP32 can in some cases, like downgrading from v2.3.x to v2.2.x, start up in AP mode without any visual indication as they do not have that LED.

gskjold, 2024-04-08:
It would be solid blue in v2.3, but yes, it does light up. I would also go to the status page and back up the configuration before upgrading.

ArnieO, 2024-04-08:
@Hawkeye4060 : You could also try the other upgrade option: Download the *.bin file to local computer and upload from there. In some cases the direct upgrade from internet fails, maybe due to lags in the connection.

Hawkeye4060, 2024-04-08:
@gskjold and @ArnieO - I have now tried again, and this time it worked flawless. 
I can't tell you what went wrong or if it went into AP-mode, but nevertheless it is now updated an working 👍 
Thank you for your continues work and fine support.

God aften fra Danmark 😉
/Mogens

closed 2024-04-08 by ArnieO